rogerclarkmelbourne/STM32duino-bootloader #3: Post bin files for PB4/PB3 for compatibility with Naze32 Board

kk6axq, 2016-01-29:
Could you compile and post bin files for a generic target with no button and and led on PB4 and PB3? This would allow compatibility with the Naze32 line of boards that run on the STM32F103CBxx series of chips.
Thanks,

rogerclarkmelbourne, 2016-01-29:
OK

I'll try to do it later today

kk6axq, 2016-01-29:
The Naze32 Acro board does not have any onboard buttons. There are two leds on the board they are both connected to VCC(3v3) then through a 100 ohm resistor to pins PB3 and PB4. So to power the leds you need to bring the pin low. Is a button absolutely necessary? I could wire one up to my board...

For reference, here's the schematic:[https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/afrodevices/naze32_rev4.pdf](https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/afrodevices/naze32_rev4.pdf)

rogerclarkmelbourne, 2016-01-29:
The button is not mandatory, only a few boards have the button.

So I guess if the board does not have a user button, it's best if I set the config for this board so it doesn't need or support a button.

Btw the button just locks the bootloader in DFU upload, and should generally not be needed

kk6axq, 2016-01-29:
Yes I think you are right about setting the config. Thanks for supporting your project!

rogerclarkmelbourne, 2016-01-30:
OK
I've added a new makefile target (make naze32) and added the config  TARGET_NAZE32 in config.h with LED on PB3 active state 0 (i,e active low), no button has been defined for this target.

Please test and let me know if it works

kk6axq, 2016-01-30:
When uploading from Windows 7 through the Arduino IDE per the wiki, I get this error:
maple_loader v0.1
Resetting to bootloader via DTR pulse
Reset via USB Serial Failed! Did you select the right serial port?
Searching for DFU device [1EAF:0003]...
Assuming the board is in perpetual bootloader mode and continuing to attempt dfu programming...

dfu-util - (C) 2007-2008 by OpenMoko Inc.
This program is Free Software and has ABSOLUTELY NO WARRANTY

Couldn't find the DFU device: [1EAF:0003]

The Naze32 board runs with a Silicon Labs driver and the VID and PID do not match the 1EAF:0003. Is there a way to manually specify the VID and PID or is there another way this can be remedied?

rogerclarkmelbourne, 2016-01-30:
Hi
The USB DFU device created by the bootloader, implements the generic DFU protocol, but is enumerated on USB as the 1EAF:0001 (I think) VID because there is no such thing as a generic VID / PID pair.

When the sketch is uploaded and starts to run, the bootloader no longer runs, and instead the sketch forces re-enumeration of the USB bus, by briefly pulling PA12 low (which then floats high again because its normally pulled up via a 1.5k resistor) I think PA12 is USB DP but it could be USB DM (i.e its one of the two USB Data lines)

The sketch then implements a generic virtual serial device, and re-enumerates as 1EAD:0003

So to use the bootloader and the serial device on windows, you have to install the drivers that effectively just associate the VID/PID numbers with WIndow's own built in drivers for DFU and Virtual serial.

It would be possible to change the VID / PID in the bootloader and the sketch to use different codes, but I doubt that this would work with the Silicon labs drivers even if the board pretended to be a sil labs device, as Sil labs probably use their own USB protocol

kk6axq, 2016-01-30:
So is there a driver I should be using for the board?

rogerclarkmelbourne, 2016-01-31:
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/drivers/win

OSX and Linux don't need drivers.  

Note the bootloader was designed to be used with the Arduino IDE to upload sketches, but I know at least one other organization which is using the bootloader to upload something completely different (as they wanted to know how to change the address the code is run from)

kk6axq, 2016-01-31:
My original problem was that I didn't flash it correctly. It's flashed correctly now from what I can tell, so when I try to upload, I now get this error:```

Arduino: 1.6.7 (Windows 7), Board: "Generic STM32F103C series, STM32F103CB (20k RAM. 128k Flash), STM32duino bootloader"

Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
In file included from C:\Users\Bob\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple/Arduino.h:30:0,

                 from sketch\Blink.ino.cpp:1:

C:\Users\Bob\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple/wirish.h:51:28: fatal error: libmaple/stm32.h: No such file or directory

 #include <libmaple/stm32.h>

                            ^

compilation terminated.

exit status 1
Error compiling.
Invalid library found in C:\Program Files (x86)\Arduino\libraries\libmaple-master: C:\Program Files (x86)\Arduino\libraries\libmaple-master
Invalid library found in C:\Users\Bob\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM: C:\Users\Bob\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM
Invalid library found in C:\Program Files (x86)\Arduino\libraries\libmaple-master: C:\Program Files (x86)\Arduino\libraries\libmaple-master
Invalid library found in C:\Users\Bob\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM: C:\Users\Bob\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

```
Any ideas?

rogerclarkmelbourne, 2016-01-31:
IDE version 1.6.7 is not supported, as it was not backwards compatible for existing users on pervious versions.

please use 1.6.5

See the readme etc

also, i presume you followed the installation imstructions in the wiki and installed the ARM compiler by imstalling a SAM or SAMD based arduino board in the boards manager

btw. this sort if issues is best resolved by the forum www.stm32duino.com as its not a bug

closed 2016-02-05 by rogerclarkmelbourne