rogerclarkmelbourne/STM32duino-bootloader #6: DFU not works

night-ghost, 2016-03-25:
on my chinese board this bootloader (tested generic_boot20_pc13.bin, maple_mini_boot20.bin, maple_rev3_boot20.bin and maple_rev5_boot20.bin) don't works: dfu_util --list can't see any DFU devices. Original LeafLabs' bootloader works fine.

rogerclarkmelbourne, 2016-03-25:
The code does not work with every STM32F103 board.

Please read the readme.md file and if necessary modify your board to meet the hardware requirements with this bootloader if you want to use it.

night-ghost, 2016-03-26:
I see you close issues without reading them...  Which "hardware requirements" in Readme.md? 

> Additionally the bootloader now works with "generic" STM32F103 boards, which do not have the additional USB reset hardware which all Maple, and Maple mini boards have.

> On "generic" boards, the USB reset (to force re-enumeration by the host), is triggered by reconfiguring USB line D+ (PA12) into GPIO mode, and driving PA12 low for a short period, before setting the pin back to its USB operational mode. This system to reset the USB was written by @Victor_pv. Note. It is not guaranteed to work on all "generic" STM32 boards, and relies on PA12 having a pull-up resistor of around 1.5k - however most "generic" boards seem to have this.

But how does this relate to the present case if original Maple bootloader works fine but this bootloader for maple doesn't?

rogerclarkmelbourne, 2016-03-26:
If it doesn't work for you. Don't use it.

rogerclarkmelbourne, 2016-03-26:
OK.

I'll put a note in the readme that it doesnt work with all STM32F103 boards

rogerclarkmelbourne, 2016-03-26:
Readme has been updated to address your concerns

night-ghost, 2016-03-26:
May be it will be better to make it does work?

night-ghost, 2016-03-28:
I did some testing. This issue is not of the board but rather  GCC 5.2 which too agressive optimized out something needed. Bootloader compiled with -O0 works fine.

c---, 2016-04-20:
Ran in to this issue myself. You have to compile with GCC 4.8 or lower. "-O0" is not really a good option because the binary is much larger. Apparently this has been a "known" issue (from the forums) for many months now. This should be in the README for sure because most Linux distros these days are using at least GCC 4.9 and it won't work. There is a possible fix that I believe involves using "volatile" for some of the variables but I don't know the specifics (there were some pointers in the forum).

Using GCC 4.9+ results in the USB device not enumerating. "unable to enumerate USB device"

ajaborsk, 2016-07-30:
Hi !
I have the same problem with **downloaded** binary. If I clone & rebuild with -O1, it works but is probably a little too big : No linker alert (one with -O0 : don't fit in flash) but it write at 0x8002000... I've a Maple mini clone, with LED on PB1, Archlinux last ARM toolchain (GCC 6.1.1...). No sketch tested for now.

Philius, 2016-08-22:
GCC can re-order instructions if it spots a chance at reducing code size or increasing performance.
For gcc __sync_synchronize is the compiler directive that tells gcc not to move stuff across it.
So you'd need to either put this between all instructions you don't want swapped around.
If it's a memory address range you're talking about then declaring the memory as volatile might do it.

Philius, 2016-09-01:
OK, now I'm typing with the hardware on a [breadboard](https://www.dropbox.com/s/0y4goflo5q8iqb3/stm32-breadboard.jpg?dl=0)
Arduino_STM32/tools/linux64/stm32flash$ ./stm32flash -k /dev/ttyUSB0
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Interface serial_posix: 57600 8E1
Failed to init device.

I get this error when trying to upload a sketch too.
How do I enable "perpetual bootloader mode" with this device.
In http://wiki.stm32duino.com/index.php?title=Blue_Pill
You mentioned connecting a resistor between PA14 and 3V3 - what value resistor?

Philius, 2016-09-03:
I compiled stm32flash from source (linux64).
In stm32.c, stm32_send_init_seq() after port->read() I added

	if (p_err != PORT_ERR_TIMEDOUT)
		fprintf(stderr, "[status 0x%02x] Got byte 0x%02x\n", p_err, byte);
	else
		fprintf(stderr, "Read timed out.\n");

I also changed the generic "Failed to init device." messages to "Failed to init device (1)." and "Failed to init device (2)." just to tell them apart.
So now I'm getting

	Arduino_STM32/tools/linux64/src/stm32flash_serial/src$ ./stm32flash -k /dev/ttyUSB0
	stm32flash Arduino_STM32_0.9
	
	http://github.com/rogerclarkmelbourne/arduino_stm32
	
	Interface serial_posix: 57600 8E1
	Read timed out.
	Failed to init device (2).

rogerclarkmelbourne, 2016-09-03:
@Philius 

Does this issue relate to the bootloader code in this repo?

Philius, 2016-09-03:
Yes, I agree without knowing what bootloader is currently installed it's difficult to know how to talk to it.

I'm trying to get started using STM32.

I tried uploading a sketch, that didn't work.

If I understand correctly uploading your bootloader is required to use your tools to upload sketches.

Do I need a smaller value resistor than 10K for PC14->3v3 to get perpetual bootloader mode?

Sometimes I get "Warning: the interface was not closed properly." so there's something on the other side replying.

Maybe it's the ST factory bootloader installed in rom but it's a guess without knowing if I've got it set up in "perpetual bootloader mode".

In serial_posix.c serial_posix_read() I added:
- a  retry_count that's incremented if read() returns 0 bytes and bails out when it reaches 100
- an errno check after read()

but it still times out.

rogerclarkmelbourne, 2016-09-04:
Hi @Philius 

Support for both the bootloader and the main STM32 Libmaple repo is on the forum

www.stm32duino.com

Philius, 2016-09-04:
Ok, so two things:

- what topic should I post to - could you provide a link?
- should I re-post what I've written so far to that topic, or create a new one, but what to call it?
"Aliexpress bluepill variant no.2" ?

rogerclarkmelbourne, 2016-09-04:
http://www.stm32duino.com/

Re post your whole question. Hardly anyone reads this repo.

Philius, 2016-09-05:
In that case may I suggest you create a top-level forum topic called "Raspberry Pi".
I found https://learn.adafruit.com/programming-microcontrollers-using-openocd-on-raspberry-pi/wiring-and-test

First of all OMG!!!

Second, I followed the tutorial and got as far as talking to the STM32F103C8T6 but I don't know how to turn of flash protect - here's the openocd output:

	RaspberryPi/stm32f103c8t6 $ sudo openocd
	Open On-Chip Debugger 0.10.0-dev-00371-g81631e4 (2016-09-04-23:07)
	Licensed under GNU GPL v2
	For bug reports, read
			http://openocd.org/doc/doxygen/bugs.html
	BCM2835 GPIO nums: swclk = 25, swdio = 24
	BCM2835 GPIO config: srst = 18
	srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
	adapter speed: 1000 kHz
	adapter_nsrst_delay: 100
	srst_only separate srst_nogate srst_push_pull connect_deassert_srst
	cortex_m reset_config sysresetreq
	srst_only separate srst_nogate srst_push_pull connect_deassert_srst
	adapter_nsrst_delay: 100
	adapter_nsrst_assert_width: 100
	Info : BCM2835 GPIO JTAG/SWD bitbang driver
	Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
	Info : clock speed 1001 kHz
	Info : SWD DPIDR 0x1ba01477
	Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
		TargetName         Type       Endian TapName            State
	--  ------------------ ---------- ------ ------------------ ------------
	0* stm32f1x.cpu       cortex_m   little stm32f1x.cpu       halted
	stm32f1x.cpu: target state: halted
	target halted due to debug-request, current mode: Thread
	xPSR: 0x01000000 pc: 0x1ffff020 msp: 0x200001fc
	openocd.cfg:17: Error: invalid subcommand "bootloader 0"
	in procedure 'script'
	at file "embedded:startup.tcl", line 60
	in procedure 'stm32f1x' called at file "openocd.cfg", line 17

Philius, 2016-09-05:
Here's my openocd.cfg with those bootloader commands commented out.:

	source [find interface/raspberrypi2-native.cfg]
	transport select swd
	
	set CHIPNAME stm32f1x
	source [find target/stm32f1x.cfg]
	
	# did not yet manage to make a working setup using srst
	#reset_config srst_only
	reset_config  srst_nogate
	
	adapter_nsrst_delay 100
	adapter_nsrst_assert_width 100
	
	init
	targets
	reset halt
	#stm32f1x bootloader 0
	program generic_boot20_pc13.bin verify
	#stm32f1x bootloader 8192
	reset
	shutdown

Is there an stm32f1x equivalent to "bootloader"?

Philius, 2016-09-06:
I finally gave in to RTFM.
To build openocd.pdf you need to

	apt-get install texinfo texlive

Then

	make docs

will get you openocd-code/docs/openocd.pdf

There's a section on stm32f1x flash commands.
Here's the openocd.cfg that worked:

	source [find interface/raspberrypi2-native.cfg]
	transport select swd

	set CHIPNAME stm32f1x
	source [find target/stm32f1x.cfg]

	reset_config  srst_nogate

	adapter_nsrst_delay 100
	adapter_nsrst_assert_width 100

	flash banks
	adapter_khz 300
	init
	targets
	reset halt
	stm32f1x unlock 0
	reset
	init
	targets
	program generic_boot20_pc13.bin verify exit 0x08000000
	stm32f1x lock 0
	reset
	shutdown

and here's the output:

	Open On-Chip Debugger 0.10.0-dev-00371-g81631e4 (2016-09-04-23:07)
	Licensed under GNU GPL v2
	For bug reports, read
			http://openocd.org/doc/doxygen/bugs.html
	BCM2835 GPIO nums: swclk = 25, swdio = 24
	BCM2835 GPIO config: srst = 18
	srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
	adapter speed: 1000 kHz
	adapter_nsrst_delay: 100
	srst_only separate srst_nogate srst_push_pull connect_deassert_srst
	cortex_m reset_config sysresetreq
	srst_only separate srst_nogate srst_push_pull connect_deassert_srst
	adapter_nsrst_delay: 100
	adapter_nsrst_assert_width: 100
	#0 : stm32f1x.flash (stm32f1x) at 0x08000000, size 0x00000000, buswidth 0, chipwidth 0
	adapter speed: 300 kHz
	Info : BCM2835 GPIO JTAG/SWD bitbang driver
	Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
	Info : clock speed 300 kHz
	Info : SWD DPIDR 0x1ba01477
	Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
		TargetName         Type       Endian TapName            State
	--  ------------------ ---------- ------ ------------------ ------------
	0* stm32f1x.cpu       cortex_m   little stm32f1x.cpu       halted
	stm32f1x.cpu: target state: halted
	target halted due to debug-request, current mode: Thread
	xPSR: 0x01000000 pc: 0x1ffff020 msp: 0x200001fc
	Info : device id = 0x20036410
	Info : flash size = 64kbytes
	stm32x unlocked.
	INFO: a reset or power cycle is required for the new settings to take effect.
		TargetName         Type       Endian TapName            State
	--  ------------------ ---------- ------ ------------------ ------------
	0* stm32f1x.cpu       cortex_m   little stm32f1x.cpu       running
	stm32f1x.cpu: target state: halted
	target halted due to debug-request, current mode: Thread
	xPSR: 0x01000000 pc: 0x1ffff020 msp: 0x200001fc
	** Programming Started **
	auto erase enabled
	wrote 7168 bytes from file generic_boot20_pc13.bin in 0.456441s (15.336 KiB/s)
	** Programming Finished **
	** Verify Started **
	verified 7104 bytes in 0.113342s (61.209 KiB/s)
	** Verified OK **
	shutdown command invoked

Now when I set BANK0 to off I get power led on but the other led flashes about twice a second.
dfu-util doesn't see the device when I connect it to the PC.

rogerclarkmelbourne, 2016-09-06:
Phillip

You really need to post this to the forum

What you are describing is not and issue with the bootloader code and should not be posted here

There are various issues why your board may not be working, Some of which are hardware related e.g. did you check the value of the USB pullup resistor ? As often these boards have incorrect values fitted.

Please register on the forum and read the FAQ

Philius, 2016-09-07:
http://www.stm32duino.com/viewtopic.php?f=37&t=1388

closed 2016-09-07 by rogerclarkmelbourne