rogerclarkmelbourne/STM32duino-bootloader #8: Using Serial UART1

tinwinaung, 2016-05-27:
Hi Roger,

I can use bootloader with china stm32f103rbt6 (https://world.taobao.com/item/38857645794.htm?fromSite=main)

I can upload scratch via USB port now.
But I also want to use existing Serial UART1 that I used to upload scratch before.

How can I use it in Ardunio code?
I try using Serial.print (and) USBSerial serial_debug;debug.print but all come out from maple_mini come port.

(I try to register http://www.stm32duino.com/index.php to ask this question but I did not receive activation email)

regards,
TIN

tinwinaung, 2016-05-27:
Found out i need to use the following
Serial1.println("OK");

rogerclarkmelbourne, 2016-05-28:
Hi @tinwinaung 

Yes.

When using the bootloader upload selection

Serial is USB Serial
Serial1 is the first hardware serial
Serial2 is the second hardware serial 
etc

If you are not using the bootloader (e.g. USB to Serial for upload) then Serial is hardware serial1

rogerclarkmelbourne, 2016-05-28:
PS.

What is your username on the forum. I will manually activate for you..

rogerclarkmelbourne, 2016-05-30:
I have manually activated you account

tinwinaung, 2016-06-02:
Hi Roger,

Sorry for late reply. Yes, I can login now.

closed 2016-06-02 by tinwinaung