rogerclarkmelbourne/STM32duino-bootloader #10: Supported boards

Philius, 2016-08-22:
Is my board supported?
How far has anyone progressed in using the specific features of this board?
These and other questions could be answered by users finding this issue or in the Wiki.

Anyway let me start the ball rolling - the description for my boards has STM32F103C8T6.
[Here's the link to the item:](http://www.aliexpress.com/item/1pcs-STM32F103C8T6-ARM-STM32-Minimum-System-Development-Board-Module-For-arduino/32478120209.html).

It might be better to embed an image in case the seller updates or stops selling the product, but I don't know how to do that.

@rogerclarkmelbourne have you made a YouTube video showing how to upload the STM32duino-bootloader onto these things?
I've got software development experience but I'd feel better with a video showing how it's done.
If not then I might give it a go providing a video for a similar board exists that I can base it off.
[This one looks like it:](https://www.youtube.com/watch?v=krxeYUiBZNY#t=407.085503)
Is flash protection on for this board? Can I disable flash protection without the STLink dongle?

rogerclarkmelbourne, 2016-08-22:
Hi

All these questions are better supported on the forum  www.stm32duino.com  as they are not really issues

Philius, 2016-08-22:
I just registered, thanks.
I'll write an intro thread and continue from there...

closed 2016-08-22 by Philius