rogerclarkmelbourne/STM32duino-bootloader #13: blupill and linux

MauroMombelli, 2016-11-14:
HI!

I have a strange issue, i'm using generic_boot20_pc13.bin downloaded the nov 13 13:04 on a bluepill.
AFTER uploading the first sketch, the AsciiTable example, every time i connect the USB dfu-util show 3 DFU but all UNKNOW and the upload will not work.

I then start to manually reset until eventually i get a /dev/ttyACM0; all the following upload of the code will work flawlessy.

dmesg

`usb usb3-port2: unable to enumerate USB device                                                                                                                                
[13887.399331] usb 3-2: new full-speed USB device number 102 using xhci_hcd
[13887.512685] usb 3-2: device descriptor read/64, error -71
[13887.739412] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13887.959388] usb 3-2: new full-speed USB device number 103 using xhci_hcd
[13888.072671] usb 3-2: device descriptor read/64, error -71
[13888.292698] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13888.512685] usb 3-2: new full-speed USB device number 104 using xhci_hcd
[13888.512804] usb 3-2: Device not responding to setup address.
[13888.719422] usb 3-2: Device not responding to setup address.
[13888.926000] usb 3-2: device not accepting address 104, error -71
[13889.229327] usb 3-2: new full-speed USB device number 106 using xhci_hcd
[13889.342748] usb 3-2: device descriptor read/64, error -71
[13889.566082] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13889.786067] usb 3-2: new full-speed USB device number 107 using xhci_hcd
[13889.899334] usb 3-2: device descriptor read/64, error -71
[13890.119345] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13890.339345] usb 3-2: new full-speed USB device number 108 using xhci_hcd
[13890.339458] usb 3-2: Device not responding to setup address.
[13890.546123] usb 3-2: Device not responding to setup address.
[13890.752662] usb 3-2: device not accepting address 108, error -71
[13890.866010] usb 3-2: new full-speed USB device number 109 using xhci_hcd
[13890.866122] usb 3-2: Device not responding to setup address.
[13891.072821] usb 3-2: Device not responding to setup address.
[13891.279331] usb 3-2: device not accepting address 109, error -71
[13891.279356] usb usb3-port2: unable to enumerate USB device                                                                                                                                
[13893.209352] usb 3-2: new full-speed USB device number 110 using xhci_hcd
[13893.322687] usb 3-2: device descriptor read/64, error -71
[13893.542733] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13893.762689] usb 3-2: new full-speed USB device number 111 using xhci_hcd
[13893.876048] usb 3-2: device descriptor read/64, error -71
[13894.096040] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13894.316032] usb 3-2: new full-speed USB device number 112 using xhci_hcd
[13894.316152] usb 3-2: Device not responding to setup address.
[13894.522768] usb 3-2: Device not responding to setup address.
[13894.729351] usb 3-2: device not accepting address 112, error -71
[13895.032687] usb 3-2: new full-speed USB device number 114 using xhci_hcd
[13895.146024] usb 3-2: device descriptor read/64, error -71
[13895.366035] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13895.586021] usb 3-2: new full-speed USB device number 115 using xhci_hcd
[13895.699421] usb 3-2: device descriptor read/64, error -71
[13895.919424] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13896.139409] usb 3-2: new full-speed USB device number 116 using xhci_hcd
[13896.139577] usb 3-2: Device not responding to setup address.
[13896.346216] usb 3-2: Device not responding to setup address.
[13896.552677] usb 3-2: device not accepting address 116, error -71
[13896.666023] usb 3-2: new full-speed USB device number 117 using xhci_hcd
[13896.666135] usb 3-2: Device not responding to setup address.
[13896.872837] usb 3-2: Device not responding to setup address.
[13897.079364] usb 3-2: device not accepting address 117, error -71
[13897.079395] usb usb3-port2: unable to enumerate USB device                                                                                                                                
[13897.566051] usb 3-2: new full-speed USB device number 118 using xhci_hcd
[13897.679344] usb 3-2: device descriptor read/64, error -71
[13897.899403] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13898.119344] usb 3-2: new full-speed USB device number 119 using xhci_hcd
[13898.232759] usb 3-2: device descriptor read/64, error -71
[13898.452702] usb 3-2: device descriptor read/64, error -71                                                                                                                                 
[13898.676009] usb 3-2: new full-speed USB device number 120 using xhci_hcd
[13898.676108] usb 3-2: Device not responding to setup address.
[13898.886136] usb 3-2: Device not responding to setup address.
[13899.092697] usb 3-2: device not accepting address 120, error -71
[13899.396024] usb 3-2: new full-speed USB device number 122 using xhci_hcd
[13899.509392] usb 3-2: device descriptor read/64, error -71
[13899.729397] usb 3-2: device descriptor read/64, error -71
[13899.949414] usb 3-2: new full-speed USB device number 123 using xhci_hcd
[13900.062694] usb 3-2: device descriptor read/64, error -71
[13900.716008] usb 3-2: new full-speed USB device number 124 using xhci_hcd`

rogerclarkmelbourne, 2016-11-15:
Try going back a few revisions and see if the older binaries work

Are you running Linux ? 

The biggest change recently was to change to use a newer version of GCC, and we thought the only issue was that it failed to jump to the sketch

But there may be some other changes which didn't effect Windows

The most recent changes have been to reduce the time taken after the USB reset is received before the sketch runs. However your issue doesnt sound like its related to that change.

You could try making the bootloader using GCC ARM that the Arduino IDE uses i.e  4.8.x

 e..g 

```
make clean 
make generic-pc13
```

and see if that resolves the issue (in which case its a compiler issue ;-(

MauroMombelli, 2016-11-15:
sorry updated comment above, the tty appear AFTER the first successfull upload, not the other way around. A friend of mine has used same version on windows and seems to work fine, another tested win + xubuntu and wors fine (30 ago 2016), he gave me the binary, tomorrow i'll try to find some time for the test

MauroMombelli, 2016-11-15:
Btw on my macchine i also nave arm-none-eabi-gcc 6.2.0, what is the compiler version you had used? Which one is the one uses in the past?

rogerclarkmelbourne, 2016-11-15:
I'm a bit confused.

Do you mean that TTY does not appear until after sketch upload? but that Upload is OK.

The bootloader does not contain any TTY code, it us only a bootloader and hence only enumerates as a DFU device

The TTY code is provided by the Arduino "sketch" (not the bootloader), because its is also possible to upload the Arduino sketch using STLink or JLink, and still have TTY (USB CDCACM)

MauroMombelli, 2016-11-15:
Yes, the  first time after uploading dfu over serial, I see the dfu and
upload is fine. Then I upload a sketch (asciitable), even multiple time,
without issue. Then i disconnect the USB, reconnect, and I got the 3
"UNKNOW" DFU (instead of saying maple bootloader V1.0 blabla) and I have to
manually reset +upload until I got it working. After that, upload works as
long as I don't disconnect the USB (which is funny as I can see the USB get
enumerated every upload...)

rogerclarkmelbourne, 2016-11-15:
Does the same problem occur if you run a simple "blink" sketch ?

(Or blink + count )

MauroMombelli, 2016-11-15:
Yes

MauroMombelli, 2016-11-18:
finally got some time, tested the version from 30 ago 2016 but is even worse, after uploading a sketch using the serial i don't t get the enumeration, and the blink does NOT blink despite the upload and reset getting reposrted as successfull!

rogerclarkmelbourne, 2016-11-18:
This may be a hardware / OS issue

Probably best to register and post on www.stm32duino.com where more people can help you.

Very few people Watch this repo

rogerclarkmelbourne, 2016-12-29:
Closed as this issue should be taken up on the forum

closed 2016-12-29 by rogerclarkmelbourne