rogerclarkmelbourne/STM32duino-bootloader #15: Add support for generic-pa8 board

gknauf, 2017-01-19:
Find attached a patch against recent git repo which adds a generic-pa8 bootloader target.
I've tested the resulting bootloader with my [cheap STM32F103RCT6 based board](https://www.fasttech.com/p/6072600).
USB upload and USB serial in/out works fine so far.

[generic-pa8.diff.txt](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/files/715465/generic-pa8.diff.txt)

rogerclarkmelbourne, 2017-01-19:
Please can you submit this as a Pull Request as I don't have a reliable way on Windows to apply patch files

gknauf, 2017-01-19:
Hi Roger,
a pull request would require me to do a fork AFAIK, and having another fork for such a simple add is a bit nasty IMO.
You can just use the Git bash:
git apply patchfile
I've just tested this against a freshly cloned repo, and it applies the patch fine but give few warnings about trailing spaces - you can ignore these; however they are correct since config.h has a bunch of trailing spaces, and since I did just copy & paste from the generic-pa9 target I produced some new ones ;-)
Also I did read somewhere that you use MinGW, and that has diff & patch utils;
so from MSYS prompt you can just do:
patch -p0 -i patchfile
should also just work fine from top level folder ... (provided patchfile and repo have same line endings)

rogerclarkmelbourne, 2017-08-20:
Things have moved on a lot since this issue was raised.
The build process has changed.

I will close this and perhaps you can create a PR which includes this change and also your other issue

closed 2017-08-20 by rogerclarkmelbourne