rogerclarkmelbourne/STM32duino-bootloader #17: Remove make_all.bat and add funtionality to Makefile

gknauf, 2017-01-24:
Since make_all.bat does not work on Linux this funtionality should be part of the Makefile so that it works on every build platform.

[all-binaries.diff.txt](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/files/725370/all-binaries.diff.txt)

rogerclarkmelbourne, 2017-01-24:
Thanks.

Looks like a good solution.

I'll see if I can update the makefile by hand, as patch doesn't seem to work very well for me on Windows.

rogerclarkmelbourne, 2017-08-20:
Things have not got a bit more complicated.

Because people did not understand that the bootloader is a DFU only USB device, I constantly got questions about why there was no Arduino Serial device

To resolve this, I have had to create a windows exe to merge the DFU bootloader with a sketch binary, so that after they install the bootloader, it actually runs a sketch which appears as a USB Serial device

Although I have published source to the program that merges the binaries, I do not use linux as my main dev platform, hence I have not compiled it for Linux (though other people have told me it does compile and run OK), and I have not written the linux shell scripts to merge the binaries

I suspect this can be done by the makefile, but I am not expert enough in writing makefiles to do this.

Please can you take a look at the latest built system and send a PR for how to do this just using the makefile and also include precompiled binaries or the various linux distros

EntropyWizard, 2017-12-17:
I looked at the merge program and think I understand what it does. It creates a merged file consisting of the boot loader code, padded to 8192 (0x2000) bytes in length and then concatenates the sample sketch to the end of the padded file. This can be accomplished on the linux command line with the following (long paths removed for clarity):
```bash
truncate -s 8192 generic_boot20_pc13.bin
cat congratulations_sketch.ino.generic_stm32f103c.bin >> generic_boot20_pc13.bin
```
I tried this on a test build of the generic bootloader and it seems to be working fine.

rogerclarkmelbourne, 2017-12-17:
I run Windows, hence I wrote my own simple program to do the same job

I was thinking of making my binary merge program a bit more generic, but time is the missing ingredient

BlackBrix, 2017-12-23:
b.t.w.
you can use the "make_all.bat" under linux as well,
just insert at the first line of the file:
#!/bin/bash

then enable the file to be executable:
chmod +x make_all.bat

the execute it with:
./make_all.bat

thats all ...

rogerclarkmelbourne, 2019-02-18:
Ideally there would be a "all" target for the make file. 

If someone can write this and submit a PR I'll merge it

closed 2019-02-18 by rogerclarkmelbourne