rogerclarkmelbourne/STM32duino-bootloader #21: How to flash bootloader with ST-Link dongle?

latonita, 2017-04-09:
I saw several mentions that it is "easy to flash bootloader with Serial or ST-Link" dongles, but instructions that i found are for serial mode only. I do have generic 'blue pill' with stm32 f103c8.

It is not clear for me still how to flash bootloader with ST-Link dongle. Can you please point me to instruction if any? Thanks a lot!

For some weird reason I don't have access to my usb-uart dongles for some time. :)

stevstrong, 2017-04-09:
Use the [STM32 ST_Link Utility](http://www.st.com/en/embedded-software/stsw-link004.html) to burn the blue pill bootloader.

latonita, 2017-04-09:
Thanks a lot!

stevstrong, 2017-04-11:
Can you please close the issue if it is solved?

closed 2017-04-12 by latonita

LightningStalker, 2017-11-03:
So stm32flash on Linux then?