rogerclarkmelbourne/STM32duino-bootloader #24: Add support for generic-pb11 board

alexxy, 2017-07-29:
Hi!

There is cheap board from aliexpress [STM32F103RCT6](https://aliexpress.com/item/STM32F103RCT6-ARM-STM32-Minimum-System-Development-Board-Module-For-arduino-Minimum-System-Board-STM32F103C8T6-upgrade-version/32785231499.html?spm=a2g0s.9042311.0.0.tMemHY)

Can you add build target for this board?

Led is on PB11

rogerclarkmelbourne, 2017-07-30:
Just use something like generic PC13, it should work, but the LED won't flash

Take a look in config.h, 

https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/STM32F1/config.h#L148-#L158

You can add your own section for with definitions you require then update the Makefile with a new target

https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/STM32F1/Makefile

Send me a PR when its done. Preferably including the compiled binary in its raw state and also combined with the blank sketch

alexxy, 2017-08-01:
Ok. Actualy i already test it with pc13 or something similar, it works but without led =) I tested blink sketch to find out on what pin led is.

rogerclarkmelbourne, 2017-08-20:
Basically I don't have time to add all possible combinations of LED pin and button.

If you would like to add your board to the build config, please submit a PR for this

closed 2017-08-20 by rogerclarkmelbourne