rogerclarkmelbourne/STM32duino-bootloader #26: Combine STM32F1 and GD32F1

GrantM11235, 2017-08-01:
Because the STM32F1 and GD32F1 folders are so similar, I think it makes sense to combine them. I have made a branch for each in the hopes of eventually merging them together. I have gotten the diff down to three files: Makefile, config.h, and hardware.c. Makefile and config.h should be pretty strait forward to merge, but hardware.c looks like it could get messy.

Check out the branches and tell me what you think:
[stm32f1](https://github.com/GrantM11235/STM32duino-bootloader/tree/stm32f1)
[gd32f1](https://github.com/GrantM11235/STM32duino-bootloader/tree/gd32f1)

rogerclarkmelbourne, 2017-08-01:
I agree

There is very little difference in terms of the bootloader.

From what I can remember, only the PLL multipler would need to be changed, as the GD32 boards I have use a 12Mhz crystal instead of the normal 8Mhz one.

GrantM11235, 2017-08-02:
I copied setupCLK() in hardware.c from STM32F1 to GD32F1, can you check that [this commit](https://github.com/GrantM11235/STM32duino-bootloader/commit/907db1b4bc1f05471e56442ebee124747e9ef758) looks right? It looks like PLL for the GD32F1 branch was set to 108MHz, is there any conflict if it is set to 72MHz like in the STM32F1 branch? It should be functionally equivalent otherwise.

rogerclarkmelbourne, 2017-08-02:
72Mhz for both is fine.

I don't think 108Mhz works, as there isnt a matching USB PLL divider to make 48 Mhz

AFIK the only clock speeds that support USB are 48Mhz, 72Mhz, 96Mhz and 120Mhz

GrantM11235, 2017-08-02:
Actually, it wasn't at 108MHz before, it was at 96MHz, the comment I was reading was outdated. I guess that doesn't really matter now that it is replaced though.

rogerclarkmelbourne, 2017-08-20:
This is been done

closed 2017-08-20 by rogerclarkmelbourne