rogerclarkmelbourne/STM32duino-bootloader #30: Generic Target Button

cdrose, 2017-08-28:
There is a line in the readme regarding the button pin that is a bit confusing:

> Note. Most "generic" STM32F103 boards only have a reset button, and not a user / test button. So the bootloader code always configures the Button input pin as PullDown, hence if a button is not present on the Button pin (Default is **PC14**), the pin should remain in a LOW state, and the bootloader will assume that the Button is not being pressed.
> 
> IMPORTANT. If you have a board where you have external hardware attached to pin **PC12** which will pull this pin HIGH, you will need to make a new build target for you board which uses a different pin for the Button, or completely modify the code so that the Button is ignored.

Does this second paragraph refer to the button pin on PC14 or is PC12 also special and cannot have any pull up hardware attached? 

In config.h I cant see any reference to PC12 so I am wondering if this is just a typo?

```
elif defined TARGET_GENERIC_F103_PC13

    #define LED_BANK            GPIOC
    #define LED_PIN             13
    #define LED_ON_STATE        0

    // Button (if you have one)

    #define BUTTON_BANK GPIOC
    #define BUTTON_PIN 14
    #define BUTTON_PRESSED_STATE 1
```

rogerclarkmelbourne, 2017-08-28:
I think its a typo

I"m in the process of changing the button pin, and will probably move it to Boot1 PB2 for all generic boards that don't have a button

Or I will remove that feature completely from boards that don't actually have a User button

cdrose, 2017-08-28:
I thought as much. I'm working with a custom board at the moment and haven't got the bootloader working correctly yet, just wanted to be sure that it wasn't my hardware on PC12 causing the problem.

rogerclarkmelbourne, 2017-08-28:
PA12 is the USB DP pin, which is pulled low and then released again, as a hacky way to reset the USB bus and force the host to enumerate, but AFIK PC12 does not get touched

rogerclarkmelbourne, 2018-04-30:
The button pin on the Blue Pill (Generic PC13 has been moved to Boot1)

I will need to check the other board targets to see if any of them really have a button or whether they need to be changed to use Boot1 as well

closed 2019-02-18 by rogerclarkmelbourne