rogerclarkmelbourne/STM32duino-bootloader #31: Device init failed

danielkucera, 2017-12-16:
Hi Roger,

I'm trying to make my st-link clone work. I flashed `generic_boot20_pa9.bin`, LED blinks at boot, I can read serial output from `/dev/ttyACM0` but `stm32flash` says `Failed to init device.` Can you help please?

```[47578.459504] usb 2-3: new full-speed USB device number 17 using xhci_hcd
[47578.600181] usb 2-3: New USB device found, idVendor=1eaf, idProduct=0003
[47578.600183] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[47578.600185] usb 2-3: Product: Maple 003
[47578.600186] usb 2-3: Manufacturer: LeafLabs
[47578.600187] usb 2-3: SerialNumber: LLM 003
[47585.535844] usb 2-3: USB disconnect, device number 17
[47585.839527] usb 2-3: new full-speed USB device number 18 using xhci_hcd
[47585.981137] usb 2-3: New USB device found, idVendor=1eaf, idProduct=0004
[47585.981142] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[47585.981144] usb 2-3: Product: Maple
[47585.981146] usb 2-3: Manufacturer: LeafLabs
[47585.981785] cdc_acm 2-3:1.0: ttyACM0: USB ACM device
danman@silverhorse:~$ sudo ./.platformio/packages/tool-stm32duino/stm32flash/stm32flash /dev/ttyACM0 
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Interface serial_posix: 57600 8E1
Failed to init device.

danman@silverhorse:~$ cat /dev/ttyACM0 
Congratulations, you have installed the STM32duino bootloader
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

For more information about Arduino on STM32
and http://www.stm32duino.com

Congratulations, you have installed the STM32duino bootloader
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

For more information about Arduino on STM32
and http://www.stm32duino.com

Congratulations, you have installed the STM32duino bootloader
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

For more information about Arduino on STM32
and http://www.stm32duino.com

```

GrantM11235, 2017-12-16:
`stm32flash` is used to upload using the built-in serial bootloader. To upload using the STM32Duino bootloader, use `maple_upload`. (PlatformIO puts it in `~/.platformio/packages/tool-stm32duino`)

stevstrong, 2017-12-16:
Once you flashed the bootloader, this will set the SWD pins to normal GPIO function, so that STLink cannot connect to the device using the STlink utility.
But you can still upload over USB and the just flashed Stm32duino bootloader.
If you want to use STLink for upload, then you have to first re-enable the SWD interface by adding the linker directive `-DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1` to [line 374 of the boards.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/boards.txt#L374).
Then upload the blinky sketch.
Now you can use the STLink again.

danielkucera, 2017-12-16:
ok, so how do I use stm32duino uploader in platformio (or directly) ?

GrantM11235, 2017-12-16:
Add `upload_protocol = dfu` to platformio.ini. There was also a [PlatformIO bug](https://github.com/platformio/platform-ststm32/issues/77) that was fixed recently, so you may need to change `platform = ststm32` to `platform = https://github.com/platformio/platform-ststm32.git` to use the latest pre-release version.

danielkucera, 2017-12-16:
it still doesn't work:
platformio.ini
```[env:genericSTM32F103C8]
platform = https://github.com/platformio/platform-ststm32.git
#board = genericSTM32F103C8
board = maple
framework = arduino
#upload_protocol = maple
upload_protocol = dfu
```
output:
```
Looking for upload port...
Auto-detected: /dev/ttyACM0
Uploading .pioenvs/genericSTM32F103C8/firmware.bin
No DFU capable USB device found
*** [upload] Error 1
dfu-util 0.7
 
Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org
 
Filter on vendor = 0x1eaf product = 0x0003
```

GrantM11235, 2017-12-16:
Try `board = bluepill_f103c8`

danielkucera, 2017-12-16:
still, the same... and thank you for helping me.
```Looking for upload port...
Auto-detected: /dev/ttyACM0
Uploading .pioenvs/genericSTM32F103C8/firmware.bin
Failed to open serial device.
dfu-util 0.7
 
Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org
 
Filter on vendor = 0x1eaf product = 0x0003
No DFU capable USB device found
*** [upload] Error 1
```

danielkucera, 2017-12-16:
I successfully uploaded firmware like this:
`sudo ./dfu-util/dfu-util -a 2 -D /home/danman/Projects/stm32-443rx/project/.pioenvs/genericSTM32F103C8/firmware.bin`

When I plug-in my device it starts as `1eaf:0003` and after few seconds changes to `1eaf:0004`. When it is 0003, I can use dfu-util to list upload targets:
```sudo ./dfu-util/dfu-util --list 
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Found DFU: [1eaf:0003] ver=0201, devnum=63, cfg=1, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000", serial="LLM 003"
Found DFU: [1eaf:0003] ver=0201, devnum=63, cfg=1, intf=0, alt=1, name="STM32duino bootloader v1.0  Upload to Flash 0x8005000", serial="LLM 003"
Found DFU: [1eaf:0003] ver=0201, devnum=63, cfg=1, intf=0, alt=0, name="STM32duino bootloader v1.0  ERROR. Upload to RAM not supported.", serial="LLM 003"
```
and also the upload works:
```
sudo ./dfu-util/dfu-util -a 2 -D /home/danman/Projects/stm32-443rx/project/.pioenvs/genericSTM32F103C8/firmware.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        13876 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
```

Now, how to make it work from platformio?

danielkucera, 2017-12-16:
My device is simply not resetting, even with [usbreset](https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line) utility:
```
danman@silverhorse:~/.platformio/packages/tool-stm32duino$ lsusb
Bus 002 Device 072: ID 1eaf:0004  
danman@silverhorse:~/.platformio/packages/tool-stm32duino$ sudo ./usbreset /dev/bus/usb/002/072 
Resetting USB device /dev/bus/usb/002/072
Reset successful
danman@silverhorse:~/.platformio/packages/tool-stm32duino$ lsusb
Bus 002 Device 072: ID 1eaf:0004  
```

GrantM11235, 2017-12-16:
Try running `~/.platformio/packages/tool-stm32duino/maple_upload /dev/ttyACM0 2 1EAF:0003 ~/Projects/stm32-443rx/project/.pioenvs/genericSTM32F103C8/firmware.bin`

danielkucera, 2017-12-16:
upload-reset doesn't work on my device:
```
bash -x ~/.platformio/packages/tool-stm32duino/maple_upload ttyACM0 2 1EAF:0003 ~/Projects/stm32-443rx/project/.pioenvs/genericSTM32F103C8/firmware.bin
+ '[' 4 -lt 4 ']'
+ dummy_port=ttyACM0
+ altID=2
+ usbID=1EAF:0003
+ binfile=/home/danman/Projects/stm32-443rx/project/.pioenvs/genericSTM32F103C8/firmware.bin
+ dummy_port_fullpath=/dev/ttyACM0
+ '[' 4 -eq 5 ']'
+ dfuse_addr=
+++ dirname /home/danman/.platformio/packages/tool-stm32duino/maple_upload
++ cd /home/danman/.platformio/packages/tool-stm32duino
++ pwd
+ DIR=/home/danman/.platformio/packages/tool-stm32duino
+ /home/danman/.platformio/packages/tool-stm32duino/upload-reset /dev/ttyACM0 750
setRTS(): TIOCMSET: Protocol error
Failed to open serial device.
```
even if I run it as sudo:
```
sudo /home/danman/.platformio/packages/tool-stm32duino/upload-reset /dev/ttyACM0 750
setRTS(): TIOCMSET: Protocol error
Failed to open serial device.
```

danielkucera, 2017-12-16:
even if it sometimes finishes successfully, it doesnt reset the device: 
```
sudo /home/danman/.platformio/packages/tool-stm32duino/upload-reset /dev/ttyACM0 750 ; echo $? ; lsusb
0
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 8087:07dc Intel Corp. 
Bus 002 Device 004: ID 0bda:572a Realtek Semiconductor Corp. 
Bus 002 Device 102: ID 1eaf:0004  
Bus 002 Device 002: ID 04f3:024d Elan Microelectronics Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

danielkucera, 2017-12-16:
For now, I've added following to maple_upload and I plug it off and in manually:
```
"${DIR}/upload-reset" ${dummy_port_fullpath} 750
echo "Waiting for device"
while ! lsusb | grep -i ${usbID}
do
  sleep 1
  echo "."
done
```

danielkucera, 2017-12-16:
It is very possible that my device doesn't have pull-up resistor from USB D+ to VCC as described here:
https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/README.md

GrantM11235, 2017-12-16:
It looks like PlatformIO uses an outdated (unknown) version of the tools from <https://github.com/rogerclarkmelbourne/Arduino_STM32>. You could try copying the contents the appropriate tools folder to `~/.platformio/packages/tool-stm32duino` to see if that improves anything.

rogerclarkmelbourne, 2017-12-17:
This discussion seems like it migrated well outside the scope of it being a possible issue on the bootloader code and is something to do with Platformio.

closed 2017-12-17 by rogerclarkmelbourne

varun-r-mallya, 2026-03-08:
> This discussion seems like it migrated well outside the scope of it being a possible issue on the bootloader code and is something to do with Platformio.
@rogerclarkmelbourne 

Please let issues continue being discussed. If you really want to close it, please tell people to continue the discussion right here. I am having trouble with this and I saw that you closed all discussions on this on very similar grounds which is giving me a lot of issues :(