rogerclarkmelbourne/STM32duino-bootloader #34: Wrong config for maple rev5

bula87, 2018-02-06:
According to schematic (https://github.com/leaflabs/maple/blob/master/maple-r5/maple-r5-schematic.pdf) USB DISC pin is connected to PA12 not PB9.

Regards
Wojciech Koprowski

BlackBrix, 2018-02-06:
this issue is the subject of a pull-request already:
--> https://github.com/rogerclarkmelbourne/STM32duino-bootloader/pull/20

rogerclarkmelbourne, 2018-04-30:
Umm.

OK. 

Looks like this has been overlooked

I'll try to get around to changing the code, but I"ll need to manually merge the PR as the code base has moved on

ngmacha, 2018-06-20:
Are binaries updated with this fix? The maple_rev5_boot20.bin downloaded from binaries folder doesn't work on my original LeafLabs' Maple rev.5 and I have to use the Olimexin's version, which is 70k... Would love to have a bit more memory with this one, but don't have the environment to compile from source.
Thank you.

yoh-there, 2019-01-02:
Bumping this issue. I can confirm [and correct *) ] this issue.
- both the schematics (Maple R5 and Olimexino-STM32 (clone) state PC12 as DISC pin
- this is not in the master branch as of today
- binary does not work on my Olimexino, so probably fair to say not in the maple_r5 binary either

I have not rebuild but will as soon as possible.

*) The correction is the original issue #20 correctly states PC12, but up-thread it states PA12, which I believe s not correct.

dpavlin, 2019-01-21:
PC12 is correct. I rebased changes on top of master, so we can merge it now so it doesn't get stale again.

yoh-there, 2019-01-21:
Thank you @dpavlin Can I suggest to also include #define BUTTON_PRESSED_STATE 1 for the maple R5?

dpavlin, 2019-01-21:
@yoh-there thanks, I did.

yoh-there, 2019-01-21:
Pffffff, I am so sorry. I must go blind ;-( Thanks again.

closed 2020-05-07 by rogerclarkmelbourne