rogerclarkmelbourne/STM32duino-bootloader #37: DFU Mode not working on generic board

oskarn97, 2018-02-28:
Hey!
Thanks for the cool project.
I am trying to use it to enable Program download via USB in DFU Mode.
I have a generic board with a 1k5 Pull Up resistor and no additional reset hardware.

When plugging in, I get the attached messages and no DFU device is being found.
Thanks a lot in advance!

`Feb 28 16:08:01 debug kernel: [ 9685.451410] usb 1-1.2: new full-speed USB device number 9 using dwc_otg
Feb 28 16:08:01 debug kernel: [ 9685.551436] usb 1-1.2: device descriptor read/64, error 18
Feb 28 16:08:02 debug kernel: [ 9685.771403] usb 1-1.2: device descriptor read/64, error 18
Feb 28 16:08:02 debug kernel: [ 9685.991399] usb 1-1.2: new full-speed USB device number 10 using dwc_otg
Feb 28 16:08:02 debug kernel: [ 9686.091420] usb 1-1.2: device descriptor read/64, error 18
Feb 28 16:08:02 debug kernel: [ 9686.311414] usb 1-1.2: device descriptor read/64, error 18
Feb 28 16:08:02 debug kernel: [ 9686.541397] usb 1-1.2: new full-speed USB device number 11 using dwc_otg
Feb 28 16:08:03 debug kernel: [ 9686.981404] usb 1-1.2: device not accepting address 11, error -32
Feb 28 16:08:03 debug kernel: [ 9687.081414] usb 1-1.2: new full-speed USB device number 12 using dwc_otg
Feb 28 16:08:03 debug kernel: [ 9687.521423] usb 1-1.2: device not accepting address 12, error -32
Feb 28 16:08:03 debug kernel: [ 9687.521558] usb 1-1-port2: unable to enumerate USB device`

Best
Oskar

BlackBrix, 2018-02-28:
which MCU ?,
which binary-file did you flash,
or have you compiled on your own with different parameters (if yes which parameters) ?

oskarn97, 2018-02-28:
STM32F103CBT6
bootloader_only_binaries/generic_boot20_pb0.bin

oskarn97, 2018-02-28:
Seems to work with precompiled binary. However, I need a custom build...

rogerclarkmelbourne, 2018-04-30:
Its open source

Its up to you to make a "Custom build"

closed 2018-04-30 by rogerclarkmelbourne