rogerclarkmelbourne/STM32duino-bootloader #38: binaries vs bootloader_only_binaries

lucky62, 2018-03-10:
What is the differrence between  `binaries`  and  `bootloader_only_binaries`?

lucky62, 2018-03-10:
I got it.
Probably it will be better to rename the directories like this:
`binaries` -> `binaries_with_welcome_sketch`
`bootloader_only_binaries` -> `binaries`

If I correctly understand, the wellcome sketch is replaced by any sketch first time burned to board. Yes?

nostromo123, 2018-03-24:
Yes, I think so.

The "welcome sketch" just gives a message in Arduino IDE terminal

_Congratulations, you have installed the STM32duino bootloader
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

For more information about Arduino on STM32
and http://www.stm32duino.com_

But this "welcome sketch" is overwritten first time you upload your own sketch.

closed 2018-04-30 by rogerclarkmelbourne