rogerclarkmelbourne/STM32duino-bootloader #40: Using the STM32 core also for the bootloader

dazhbog, 2018-04-10:
Hi there,

Apart from space constrains, are there other reasons for not using the STM32duino core for the bootloader?

If we use the arduino core in the bootloader and in the application then technically we could have CDC serial on the bootloader to do firmware updates and then jump to user code without needing to re-enumerate? Or is it because in application memory the vector table is re-allocated after the bootloader, CDC needs to be re-enumerated?

Another benefit is that would make it easier to handoff application code to the bootloader. In some cases when we want to add existing radio or external flash code or i2c from the application side to the bootloader to implement OTA or load FW from external flash on boot.

Im thinking of giving this a go, are there any other pitfalls or issues with doing this?

stevstrong, 2018-04-10:
Isn't this rather an issue of the SDM23DUINO core instead of the bootloader?
The core should decide about the bootloader and not vice versa.

dazhbog, 2018-04-10:
Could be as well yes. 

The reason I put it here is to learn what are the constrains that stop us from getting to the _result_, which is a more easy to work with bootloader.

rogerclarkmelbourne, 2018-04-30:
The bootloader is unrelated to the core.

You can use the bootloader with any core.

Combining the bootloader into the core would potentially cause a lot of problems.

Cypress tried that with their original PSOC product line, but if there is a bug in the resultant sketch the bootloader failed to run and you had to re-install the bootloader.

If you want a different sort of bootloader, take a look at the work @jcw has done (see JeeLabs)
However I think he abandoned the bootloader as well, due to similar problems.

closed 2018-04-30 by rogerclarkmelbourne

jcw, 2018-04-30:
Yes, it's definitely _very_ tricky - as you say, the hard case is how to get out of an upload which hangs, i.e. a fatal bug on the user's sketch. With a separate boot loader, this can be handled on reset. I _do_ think it's possible to combine things, but haven't gotten around to it yet. FWIW, my last work on this was a few years ago, see [github](https://github.com/jeelabs/embello/tree/master/tools/usbserup).

dazhbog, 2018-05-01:
Thanks for the input. Completely understand the risks and tradeoffs. 

Just to clarify, I meant to have the bootloader compiled with the stm32duino core, hence exposing gpios, sleep, and other peripherals with the familiar APIs. But absolutely the bootloader would be **separately compiled and linked from the application itself.**

Motivation for this is simple, you spend all this time implementing sleep, external spi flash, Lora, GSM, WiFi, and all these other cool things, and then you might want to put some of this _battle tested_ code onto the bootloader side (separately compiled) each with their own heap, stack, and vector tables. 

I made a really rough bootloader with the stm32duino core last weekend. It ended up being really bloated, and it seems to crash when I write on the VTOR vector table offset register. 😢 I guess even if it works though its still wasting alot of KBs and its doomed to fail.

rogerclarkmelbourne, 2018-05-01:
@dazhbog 

You should take a look at the HID bootloader which @bootsector has written and which has been modified by @Serasidis 

Its much simpler and does not need a driver, but it does need a special uploader program

If we also had a HID serial interface (which would need a separate serial terminal program), it may be possible to fool the host OS into not realizing that the bootloader had stopped running and the USB was being handled by the sketch 

However, getting this sort of thing to work seamlessly is a lot of work and well beyond the scope of this bootloader, or what I have time to do