rogerclarkmelbourne/STM32duino-bootloader #43: Compiling bootloader from scratch

TamojitSaha, 2018-04-29:
Hello Roger,
After going through the instructions in [README](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/README), I tried run `make generic-pc13` from command line in Windows. But after going through the clean process, it stops and says "_The application was unable to start correctly(0xc0000142)_". I am using **WinAVR** _make_ command to compile. I have attached a screenshot of my PC for your reference. Please help me out.
[image]

SandeepanSengupta, 2018-04-30:
If you have solved it, please post the solution here

rogerclarkmelbourne, 2018-04-30:
Use GNU make

closed 2018-04-30 by rogerclarkmelbourne

TamojitSaha, 2018-05-01:
I have written full instruction to compile bootloaders from scratch in Windows Platform.

Hope it will be useful for someone.

[Here](https://github.com/TamojitSaha/Compiling_Arduino_STM32_bootloaders) is the link.

@rogerclarkmelbourne Please verify.

TamojitSaha, 2018-05-01:
@rogerclarkmelbourne 
I suggest you add the steps to compile after @SandeepanSengupta  verifies the procedure. He is facing the same error as mine.