rogerclarkmelbourne/STM32duino-bootloader #45: MSD-CDC Composite bootloader for STM32F103

SandeepanSengupta, 2018-05-01:
Is it possible to develop a composite boot-loader like [this](https://github.com/watterott/SAM-BAR) ?
Aslo, I found [this](http://www.avislab.com/blog/stm32-bootloader_ru/) project that managed to implement a MSD partition.
USB and MSD always fascinated me as it will allow drag and drop programming like [this](https://www.youtube.com/watch?v=oWqUYnxkqAs)

uTasker, 2019-01-27:
USB-MSD loader is available for Blue Pill at http://www.utasker.com/stm32/BluePill.html

rogerclarkmelbourne, 2019-01-27:
Thats for sharing

Note. There is also now a HID bootloader, which is a bootloader which appears as a HID device and hence does not need a driver. However the HID bootloader needs a special uploader program

BTW. The downside of a Mass Storage Device bootloader is that it takes much more program space (flash) than either this bootloader or the HID bootloader.
Of the three, the latest version of the HID bootloader is by far the smallest.

uTasker, 2019-01-27:
Yes, USB-MSD is a bit more complicated and occupies 12k in Flash (this is with an OS for simplicity). USB-MSD however also doesn't require a driver and so works on Linux, MAC or PC without needing any PC program, which makes it popular.
I also have the KBOOT HID loader (takes up less space - around 8k, with OS) [renamed to MCUBOOT recently] which can be used the Blue Pill too. The KBOOT host software (blhost - https://www.nxp.com/docs/en/user-guide/MCUBLHOSTUG.pdf and https://www.nxp.com/docs/en/user-guide/MBOOTDEMOUG.pdf) is available for Linux, MAC and PC but it is supplied y NXP for their Kinetis and LPC devices so they probably won't like it being used with STM32 parts (although it does without any difficulties).
I also have AES256 encrypted versions of the two (a bit bigger of course) for when the loader is integrated in products based on the HW for code protection.
The loader in fact works on STM32F1, F2, F3, F4 and F7 parts (actually I did the Blue Pull exercise because I didn't have a USB device driver beforehand so I used this as test vehicle for the USB device driver development/verification).

rogerclarkmelbourne, 2019-01-27:
No worries.
I realise a MSD device does not need a driver.

I have seen another MSD bootloader which I think is a bit smaller but it does not implement a file system (I presume yours implements FAT32).

But there is no advantage of having a MSD bootloader unless the OS can upload the file without another application, because you may as well use the HID bootloader, which I think now may only be 4k

uTasker, 2019-01-27:
USB-MSD loading also needs no PC SW - it just uses drag-and-drop via the file manager.
Here is a version from 2011 on a SAM7: https://www.youtube.com/watch?v=e4oFBn_M5wo
but since WIN 8.1 (when the OS started copying hidden indexing files on the disk it got a bit more complicated and so revisions were necessary).
It doesn't use FAT but has to emulate FAT so that the host believes it is a real memory stick. The details are in http://www.utasker.com/docs/uTasker/uTaskerSerialLoader.pdf and for MAC OS compatibility at http://www.utasker.com/kinetis/USB-MSD-MAC.html
More general FAT emulation is described in http://www.utasker.com/docs/uTasker/uTaskerEmulatedFAT.pdf which allows applications to display collected data in form of files (or formatted ones for Excel and such) .
I just added a binary at the initial link in case anyone wants to see it all in action.

There is no optimal solution for boot loading and application development so a variety of possibilities is useful (to allow selecting the most suitable for a certain case). If space is a concern (a 12k boot loader limits the code size to 116k in the Blue Pill and a 4k loader to 124k (6.8% more)) a smaller loader is of course more important, but if 128k code may be needed then the internal ROM loader (via USB to UART adapter) also becomes the only solution. If a project is anticipated to occupy 128K of flash "already at the outset of project development" and so almost no boot loader space is available, the chip would possibly already be a bad choice because there is already a risk that the application will not fit.

uTasker, 2019-01-27:
New Blue Pill video: https://youtu.be/dq-m-Dokq7E

rogerclarkmelbourne, 2019-01-28:
Excellent video.

Re: Bootloader size.

I agree. People get hung up on bootloader size, when its not always that important.

One note about the BluePill Flash size. People have reported recently that some BluePills have been received containing the STM32F103C6 which is only 32k Flash, and these parts have been relabelled as STM32F103C8 (which should have 64k)

And.. There also appear to be some BluePill boards with a STM32 clone fitted.
These contain the CS32F103C8, but at the moment very little is known about these devices.
I've ordered 2 BluePills which had CS32F103C8 in the vendor description, and I'm hoping I should receive them from China in the next 2 weeks, so I'll have a better idea of how compatible these chips are.

(and of course there is the GD32 but AFIK none of the Chinese manufacturers use them on the BluePill)

uTasker, 2019-01-28:
Interesting about the flash sizes. I wonder what ST Micro has to say about this?
If 128k is inside it is  bonus (I set up my work for the "official" 64k but it could use the 128k when there) but if only 32k is there (that is, one gets a different chip to what is advertised) it could be a disaster.
In such a case commercial users of the board (that is, companies ordering batches to built in to products) will need to find another solution (one must at least be certain of getting the chips that were ordered!). The may need to clone the board themselves or just the processor in their design directly to secure their product - this may be the end of the direct Chinese source for anything else than for hobby work where such quirks may be fun rather that a serious business issue.  The Chinese manufactures may need to be more careful otherwise they could kill off this own business!

uTasker, 2019-01-29:
Note that it would also be possible to share a PID/VID pair for both loader and application but it would mean that both loader and application use a MSD and CDC composite configuration so that the host just things the same device is reconnecting again. This would however need a VCOM in the loader (the uTasker has this option since it also allows SREC/iHEX loading via CDC) and a dummy USB-MSD in the application, even when not used.
It makes more sense to just use two PIDs assuming a VID is available (thus two VID/PID combinations).
VIDs are very expensive - $5'000. Contact your chip manufacturer to apply for a PID using their VID since they usually offer such schemes (apply for 2 PIDs in your case!)
The uTasker project has VID/PID agreements with most semiconductor manufacturers that it supports (with exception from ST Micro - I don't remember why not). These include official Freescale, Atmel and Luminary Micro PIDs - and interestingly these companies no longer exist so these may be available for free use ;-) because I doubt that usb.org would be able to re-assign the blocks to new customers (due to existing products in the market).
For internal work there is a research VID that I understand can be used but if your not selling products and the devices remain internal to your school/university you can just use any value....

rogerclarkmelbourne, 2019-01-29:
Guys

Can you take this discussion somewhere else, as its not an issue in this DFU bootloader that its not a MSD bootloader

closed 2019-01-29 by rogerclarkmelbourne