rogerclarkmelbourne/STM32duino-bootloader #46: Document Linker script modification

ildar, 2018-05-15:
With using the bootloader the user projects must be changed accordingly. But it's not so obvious how. Documenting this in README or somewhere around sounds natural, agree?

rogerclarkmelbourne, 2018-05-16:
See

https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/README.md

closed 2018-05-16 by rogerclarkmelbourne

ildar, 2018-05-16:
Roger, you must have misunderstood.
The matter of the issue isn't the bootloader itself but the *user projects*,
i.e. programs that users flash to the MCU at modified addresses, e.g.
0x8002000.
Can you please reconsider and reopen?

BlackBrix, 2018-05-16:
it is metioned in the document Readme.md:

>> quote begin:
...
1.) Smaller footprint - now only 8k (This became possible due to code changes by @jonatanolofsson which allowed the GCC optimisation for size flag to be used
>>
>> 2.) Additional DFU AltID upload type was added, which allows the sketch to be loaded at 0x8002000 instead of 0x8005000 (due to reduce size of the bootloader is self), Note. upload to 0x8005000 was retained for backwards compatibility.
...
quote end <<

so everything is fine, IMHO

rogerclarkmelbourne, 2018-05-16:
Yes. Its fully documented.

And, what people upload with the bootloader is out of scope for the bootloader its self.