rogerclarkmelbourne/STM32duino-bootloader #48: Add an option to build the firmware with PlatformIO

me21, 2018-05-29:
Perhaps it's worth considering? I'd like to use the bootloader for my custom board with 12 MHz crystal, so the existing binaries don't suit me.
And I'm developing everything in PlatformIO IDE, so I'd rather not install Arduino IDE just to compile the bootloader.

ilya89vet, 2019-02-15:
Доброго времени Александр
Я если честно не понял что нужно сделать, что бы можно было загружать код в плату с боотлодером из Platformio IDE?

me21, 2019-02-15:
Если бутлоадер уже прошит, то в файле `platformio.ini` основного проекта нужно указать `upload_method = dfu`.

If the bootloader has already been uploaded to the board, you should add a line `upload_method = dfu` to your primary project's `platformio.ini` file.

ilya89vet, 2019-02-15:
upload_protocol = dfu я делал
При попытке загрузить код, плата перезагружается и определяется как
неизвестное устройство и загружаемый код не работает, после переполнения
USB плата определяется как Maple DFU. После этого приходится заново
прошивать Bootloader.

me21, 2019-02-15:
Maple DFU означает, что плата находится в режиме бутлоадера. Заново его прошивать не надо. Попробуйте на простейшем примере помигать лампочкой. Если не сработает - покажите код.

Maple DFU means that bootloader is up and running, you don't need to reflash it again. Try the simplest "LED blink" code. If it won't work please show me the code.

ilya89vet, 2019-02-16:
> Но это поддерживается только ядром Arduino авторства rogerclarkmelbourne

А как это ядро установить в PlatformIO?

me21, 2019-02-16:
Вопросы установки Arduino в Platformio - https://community.platformio.org, https://docs.platformio.org

Для использования бутлоадера надо поставить драйвер, см. https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Maple-drivers

Questions related to installing Arduino to Platformio - https://community.platformio.org, https://docs.platformio.org

You need to install driver before using bootloader - https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Maple-drivers

rogerclarkmelbourne, 2019-02-18:
The original issue is a feature request.

Subsequent posts in Russian seem to be a support request. Please read the Readme or search online about how to use the bootloader.

Issues are not a way to ask for support. There is no support. This is a free open source project, its non-commercial, hence there is no support

closed 2019-02-18 by rogerclarkmelbourne

me21, 2019-02-18:
@rogerclarkmelbourne can you please review the linked PR concerning the initial feature request?