rogerclarkmelbourne/STM32duino-bootloader #52: generic_pc13 issue

FCam1, 2018-07-13:
Hi, 
I met issues after uploading the last update of the bootloader on my STM32F103ZET6. Since this change a16ff8ed171f1905e2c160a9c1c70c8c972a52ae 
my program doesn't work anymore if  the USB is connected. With the previous version it works.

elmicha, 2018-10-26:
Thank you, FCam1 - I had the same problem. [The old version](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/d19bcaf1ef044eb5e09b772f7a1d7c158c3c2976/binaries/generic_boot20_pc13.bin) works, the current version seems to be stuck in perpetual boot loader mode.

After lots of head scratching (I'm very new to STM32, so it is very likely that I am doing something wrong), your bug report made me check if my Blue Pill's BOOT1 is actually connected to PB2.

Using it as an input didn't show anything, using it as an output didn't show anything on the volt meter.

According to the [schematic](https://wiki.stm32duino.com/images/c/c1/Vcc-gnd.com-STM32F103C8-schematic.pdf) there should be a 100kΩ resistor "R4" between the BOOT1 pin an the actual PB2.

This is how it looks on my Blue Pill:
[image]

I'm not sure if I managed to knock it off myself or if it came like this (one day I will learn to take enough pictures).

stevstrong, 2018-10-26:
I think the new version stays in bootloader mode if BOOT1 (PB2) is set to 1.
Or other way around, you have to set PB2 to 0 in order to jump to user code.

stevstrong, 2018-10-26:
As your board is not a usual one, do not expect that each special board to be supported here.
Solder an appropriate resistor if still unsatisfied. Or use the older version.

rogerclarkmelbourne, 2019-02-18:
Or just modify the bootloader to support your board

closed 2019-02-18 by rogerclarkmelbourne