rogerclarkmelbourne/STM32duino-bootloader #58: Disconnect USB after flash

JenuRudan, 2018-10-22:
The usb device is kept read as MAPLE 003, so when i initialize the usb again nothing happens, so i can't use the usb for debugging.

My firmware code

```
void MX_USB_DEVICE_Init(void)
{
  /* USER CODE BEGIN USB_DEVICE_Init_PreTreatment */
  
  /* USER CODE END USB_DEVICE_Init_PreTreatment */
  
  /* Init Device Library, add supported class and start the library. */
  USBD_Init(&hUsbDeviceFS, &FS_Desc, DEVICE_FS);

  USBD_RegisterClass(&hUsbDeviceFS, &USBD_CDC);

  USBD_CDC_RegisterInterface(&hUsbDeviceFS, &USBD_Interface_fops_FS);

  USBD_Start(&hUsbDeviceFS);

  /* USER CODE BEGIN USB_DEVICE_Init_PostTreatment */
  
  /* USER CODE END USB_DEVICE_Init_PostTreatment */
}
```

rogerclarkmelbourne, 2019-02-18:
SWD pins are disabled in the Arduino STM32 core, AFIK this is not a problem with the bootloader its self

closed 2019-02-18 by rogerclarkmelbourne

me21, 2019-02-20:
Do you use external pull-up resistor on USBDP pin? Configure this pin as output and set it to zero for ~100 microseconds, then configure USB again.
This will help USB host to detect disconnection of your device after flash.