rogerclarkmelbourne/STM32duino-bootloader #61: Compile error when the extended filename contains a blank

gerardwr, 2019-02-15:


rogerclarkmelbourne, 2019-02-18:
This is not an issue with the bootloader, its an issue with the arduino_stm32 core

Please repost the issue in that repo

closed 2019-02-18 by rogerclarkmelbourne

gerardwr, 2019-02-19:
Thanks for reply-ing and pointing me in the right direction. 

Found the same issue on arduino_stm32 core and a more simple temp fix in platform.txt
https://github.com/stm32duino/Arduino_Core_STM32/issues/256

fpistm, 2019-02-19:
> Create a new issue there for uploading:
> [stm32duino/Arduino_Core_STM32#448](https://github.com/stm32duino/Arduino_Core_STM32/issues/448)

I've answered you about this issue.

gerardwr, 2019-02-19:
@fpistm 

Ok, I'm confused, but opened a new issue here. Thanks.
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/600