rogerclarkmelbourne/STM32duino-bootloader #62: How to Run "make_all.bat"

ForeverGeeker, 2019-02-18:
I tried to install "MinGW" and tried to ran make_all.bat, but some error occurred:

`C:\Users\Administrator\Desktop\STM32duino-bootloader>make maple-mini
make: *** FATAL *** maple-mini: unknown action keyword

C:\Users\Administrator\Desktop\STM32duino-bootloader>make maple-rev3
make: *** FATAL *** maple-rev3: unknown action keyword

C:\Users\Administrator\Desktop\STM32duino-bootloader>make maple-rev5
make: *** FATAL *** maple-rev5: unknown action keyword

C:\Users\Administrator\Desktop\STM32duino-bootloader>make generic-pc13
make: *** FATAL *** generic-pc13: unknown action keyword

C:\Users\Administrator\Desktop\STM32duino-bootloader>make generic-pg15
make: *** FATAL *** generic-pg15: unknown action keyword`

Did I forget something? And I'm a high school student, trying to modify the bootloader... Can I have your reply, please?

BlackBrix, 2019-02-18:
you try to run make_all.bat within linux ?
if yes this will not work properly,
just make yourself a "makeall_linux" bash file (name doesn't matter)
with #!/bin/bash in the very first line 
see here for example:
--> https://github.com/BlackBrix/STM32duino-bootloader/blob/master/makeall_linux

ForeverGeeker, 2019-02-18:
Thanks for your help, but I try to run make_all.bat within windows...

rogerclarkmelbourne, 2019-02-18:
make_all.bat is just a convenience utility for me when I have to rebuild everything before committing back to github.

Its a windows batch file and its not intended to run on any other platforms

Its not needed if you just want to build a specific version for yourself

closed 2019-02-18 by rogerclarkmelbourne

ForeverGeeker, 2019-02-20:
I'm trying to open STM32duino-bootloader project by Keil, is it work?
If yes, the lib of stm32 show errors like
stm32_lib\cortexm3_marco.c(55-136): Invalid line start
stm32_lib\c_only_startup.s (52-54): Lable missing from line start
etc.
at about 155 error(s)
Did I make some mistakes?