rogerclarkmelbourne/STM32duino-bootloader #64: Raspberrypi 3B+ Connection Issue

theoutfield, 2019-03-13:
I noticed on the stm32duino forum you left a message about a hack for the Raspberrypi 3B+.  https://www.stm32duino.com/viewtopic.php?f=22&t=4276&sid=0fffea9b90e869ab49022c9e9fdb845e&start=10

Do you remember what the hack was?  I'm trying to get this working myself but haven't been able to.

rogerclarkmelbourne, 2019-03-13:
This project was the one using the bootloader

https://github.com/juribeparada/MMDVM_HS

I'm not sure if the bootloader is in that specific repo or if @juribeparada has a different repo with the bootloader in it.

theoutfield, 2019-03-13:
Thanks  for the quick response Roger.  I think I found it here https://github.com/juribeparada/STM32F10X_Lib/tree/1debc23063f3942608e2bd62d04d5e1249c47fa3/utils/bootloader but it didn't work.  The STM32 still shows up as leaf:003 on the Raspberrypi using the long reset bootloader.  Is it possible to not even use the bootloader?  I have an STLINK device that I can program the device with.  When I export my code from the Arduino IDE to a bin/hex file and flash it the device isn't detected as a known USB device.  I don't really need the bootloader in my case.

rogerclarkmelbourne, 2019-03-13:
You don't need to use the bootloader, the other options are USB Serial, STLink and BlackMagicProbe

However, I don't know what support there is on the RPi for these, as they all have their uploader binaries for the other platforms.

There should be source code for some of these, if you want to recompile for RPi

theoutfield, 2019-03-13:
Thanks for your help I appreciate it.  I'm a little green on this.  I was able to get the bootloader that you created installed and did my development on Windows.  When I plugged it into my PI it didn't work.  So do I just compile my code in the Arduino IDE and then export it?  How do I get the USB Serial incorporated? Is it compiled into the code?  Do I need to install a new USB driver on Linux to detect it?  Essentially I just want to use the USB to serial functionality so I can transfer data to the PI.

rogerclarkmelbourne, 2019-03-13:
Whats happening is the bootloader tried to reset the USB bus just prior to jumping to the sketch code, however the RPi 3B+ does not noticed the USB reset pulse.

If you upload via STLink on windows into the STM32, it will overwrite the bootloader with the application code, and you won't have this problem, because the reset pulse is not required, because the bootloader is not present and hence the board does not initially appear as 1eaf:0003  
Instead the board will appear straight away as 1ead:0004 which is USB Serial (CDCACM)

theoutfield, 2019-03-13:
Thanks again.  That worked.  I should have tried a little harder before asking.  I just flashed the bin/hex file directly onto the STM32 and it was detected as leaf:004 and I was getting serial data transfer.  I was thinking the bootloader had the code in it for the USB->Serial.  I guess it's included as part of the sketch when you use Serial and Linux just detected the device as such.  I'm sure your busy but if you could point me to where that gets linked into the code I wouldn't mind knowing.

rogerclarkmelbourne, 2019-03-13:
USB Serial is linked in for

Stm32duino bootloader, and STLink
Its not linked in for USB Serial upload.

I can't remember if its linked in if you select BlackMagicProbe (could be either as BMP has its own USB Serial < -- > UART)

Look in boards.txt, there is a define to say whether that option builds in USB Serial

theoutfield, 2019-03-13:
Thanks Roger.  I'm working on an electronic hockey net project for my son.  https://www.youtube.com/watch?v=POiDH1R1uNs&list=PLgCmNhrVBcXrVEuwPDwPNnpzZJ4VdgMRP  
I was a little bummed out when I couldn't connect it to the PI.  It's the last piece to finish off this project.  Now I can finish it up!

rogerclarkmelbourne, 2019-03-13:
No worries..

The RPi 3B+ has caused a lot of problems with this.

But we have no idea why they would have changed the USB handling

The RPi 3B was fine, its just the B+ (and possibly the new A+)

closed 2019-03-13 by theoutfield