rogerclarkmelbourne/STM32duino-bootloader #65: HY-MiniSTM32V bootloader?

spacerace, 2019-03-14:
Hi,

I have 2 of a HY-MiniSTM32V board with stm32f103vct6 (256k flash/64k ram/100pin). 
In another issue about an unknown board I read, that I could just use a "generic_...." bootloader image.

As this board has two LEDs on PB0 and PB1 I tried the `generic_boot20_pb0.bin` image (from both folders, binaries and bootloader_only_binaries).
Both are blinking the LED for a short time but there is no USB activity at all in my system log (dmesg).
Usually there is something like "Unable to enumerate ..." if there is something wrong with USB devices, but there is nothing...

Host system is debian linux. This board is running other USB-programs correctly (like the mass storage example from vendor), so this is not a hardware related problem.
I don't know where to start. I think it might be something about different USB circuitry on this board.

I have cutted the USB part from schematic:
[image]

And here's the full schematic if you need. 
[HY-MiniSTM32V_SCH.pdf](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/files/2968851/HY-MiniSTM32V_SCH.pdf)

I hope you can give me a hint where to start/what to do to use this board with your bootloader.

rogerclarkmelbourne, 2019-03-15:
Is Boot1 pulled low?

Try pulling Boot1 high and the bootloader should hold in DFU mode. 

Your board has a reset circuit similar to the Maple mini, so you probably need to modify the maple mini config 

https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/config.h#L95-L110

To use PB7 instead of PB9, then build for Maple mini rather than the generic board.

Its impossible to have pre-built binaries for every conceivable permutation of hardware, so you'll need to build the binary yourself.

However in theory, even for boards that have an external USB reset circuit the "generic" builds often still work OK.

In your case, it sounds a bit like the bootloader is starting up but then jumping to the main application code after a few seconds.

If you totally erase the chip and try again, it should in theory, not try to jump, because the application space should just contain 0xff ... and not a valid stack pointer location (which is what the bootloader uses to determine if there is valid application code it can jump to)

But also pulling Boot1, will confirm whether the USB connections are OK, as it should hold in DFU mode and you should be able to see the 1eaf:0003 device

PS. I think you may need to be root or change the UDEV rules in order to be able to use the USB device (1eaf:0003)

spacerace, 2019-03-15:
On my way to work, so no just a short reply.

I changed the maple config as you told me and I get this in syslog:
```
[35539.068078] usb 2-2: new full-speed USB device number 4 using uhci_hcd
[35539.244516] usb 2-2: New USB device found, idVendor=0483, idProduct=3748
[35539.244524] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[35539.244530] usb 2-2: Product: STM32 STLink
[35539.244535] usb 2-2: Manufacturer: STMicroelectronics
[35539.244540] usb 2-2: SerialNumber: QÿoP\xffffffc2\xffffff80UI9E	\xffffffc2\xffffff87
[35556.376189] usb 1-2: new full-speed USB device number 3 using uhci_hcd
[35556.550280] usb 1-2: New USB device found, idVendor=1eaf, idProduct=0003
[35556.550288] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[35556.550294] usb 1-2: Product: Maple 003
[35556.550299] usb 1-2: Manufacturer: LeafLabs
[35556.550304] usb 1-2: SerialNumber: LLM 003
```
I havent tried it with IDE uploading, short in time.
This is a step forward, thank you!

Later I will test and if everything works, I will add a new configuration to your bootloader and do a pull request.

Thanks for your help!!!

spacerace, 2019-03-18:
I've been playing around now with stm32duino and my HY-MiniSTM32V and the few things I've tested were good so far.

The problem now is, that I dont know which pin on my board is which. I think I will add this board there too.

I'll close this now and sooner or later you will receive a pull request.

Thank you very much!

closed 2019-03-18 by spacerace