rogerclarkmelbourne/STM32duino-bootloader #67: Question:  to what address should I flash a user program with ST-LINK

johncblacker, 2019-04-03:
I'm using ChibiStudio to build programs and needing to use an external st-link to flash the program onto the board.  My question is at what address should I be flashing the user program?  I see that the maple_uploader.bat would have different options: 0x08002000 or 0x8005000.  What is the correct value which I should use with st-llink?  I presume I'd have to modify the linker script, as well?

rogerclarkmelbourne, 2019-04-03:
Old bootloader was larger and the application needed to be at 0x8005000 the new bootloader is smaller and the application can be at 0x8002000

But if you are uploading directly using STLink then the bootloader is irreverent, as you can directly upload to 0x8000000 and don't need a bootloader

florisla, 2019-04-23:
@johncblacker Please close this ticket if this answers your question.

To summarize:
* Write the application to `0x08000000` if you do not use a bootloader.
* Write the application to `0x08000200` if you are using a recent bootloader.
* Write it to `0x08000500` if you are using an outdated bootloader.

cscheib, 2019-06-28:
> @johncblacker Please close this ticket if this answers your question.
>
> To summarize:
>
> * Write the application to `0x08000000` if you do not use a bootloader.
> * Write the application to `0x08000200` [...]

@florisla Sorry, corrections necessary, you need to shift the 2 and the 5 left by 1 digit.

* Write the application to `0x08000000` if you do not use a bootloader.
* Write the application to `0x08002000` if you are using a recent bootloader.
* Write it to `0x08005000` if you are using an outdated bootloader.

johncblacker, 2019-06-28:
Closing assolved.

closed 2019-06-28 by johncblacker