rogerclarkmelbourne/STM32duino-bootloader #69: The generic_boot20_pc13_fastboot.bin seems to work as an 'update'

rtek1000, 2019-04-15:
Hi

Something strange has happened with the new 'generic_boot20_pc13_fastboot.bin' bootloader.

If it is burned to the full erased chip, it will not work.

If burned on the chip with the generic_boot20_pc13.bin, after uploading some sketch, it works.

The 'generic_boot20_pc13_fastboot.bin' seems to work as an 'update' to the 'generic_boot20_pc13.bin'.

With the chip completely erased, after burn the 'generic_boot20_pc13_fastboot.bin', when connecting in Windows 10 x64, only 'libusb-win32 devices' appears without  nested device 'Maple DFU'.

For the 'generic_boot20_pc13.bin' appears the nested device 'Maple DFU' (VID_1EAF & PID_0003).

The resistor R10 is 1k5.

The 'generic_boot20_pc13.bin' starts the program earlier than 'generic_boot20_pc13.bin', so it's a good idea to do this 'update'.

But it's strange not to work with the chip full erased (by 'STM32 ST-LINK Utility').

rtek1000, 2019-04-21:
Hi,

I'm closing this issue, because I'm trying to use another bootloader, if anyone else wants to try it: https://github.com/Serasidis/STM32_HID_Bootloader

Thank you.

closed 2019-04-21 by rtek1000

raghur, 2019-06-16:
@rtek1000 - do you have detailed steps on how you flashed the fastboot as an update? I need the fastboot behavior on blue pill but flashing the binary from the repo via serial (PA9, PA10) and stm32flash serial cable does not work. 
Right now, I've flashed PC13 generic boot bin and then flashed the blink sketch from arduino IDE. How do I proceed after this?
Also, my bluepill seems to have the right R10 resistor (152)

rtek1000, 2019-06-16:
Hi, try uploading the sketch blink with the normal usb bootloader.  Then try burning fast bootloader without full erasing memory.

raghur, 2019-06-16:
>Then try burning fast bootloader without full erasing memory.

Sorry - but how? Do I use dfu-util? or something else?

rtek1000, 2019-06-16:
Note: USB bootloader + Bluepill, It may only work on Windows, due to the way the S.O. performs USB / Serial port handling.

rtek1000, 2019-06-16:
St-link V2 (Windows)

raghur, 2019-06-16:
I have the stlink dongle. I guess the texane repo is the place to look

raghur, 2019-06-16:
This worked. Thanks - FASTBOOT is really useful for battery apps. In case someone else lands here looking for more detailed instructions for bluepill:

1. Use USB to serial and flash the generic bootloader as on the wiki page https://wiki.stm32duino.com/index.php?title=Burning_the_bootloader
2. Now use arduino ide to flash blink program (not sure if it's necessary - but still)
3. Now switch over to windows, download the ST-LINK app, plugin your ST-LINK V2 and put BOOT0 to high. Download [fastboot bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/bootloader_only_binaries/generic_boot20_pc13_fastboot.bin) 
4. In st-link, click program, select the fastboot bin file, memory as 0x8000000. make sure that Erase is NOT selected.
Now you should have blue pill with the fastboot bootloader.

@rogerclarkmelbourne any chance we can get a full image that can be flashed normally (And not as update)