rogerclarkmelbourne/STM32duino-bootloader #75: generic_boot20_pc13.bin with older STM32F103C8T6 Rev X chip.

dnfi, 2019-09-04:
I have 5 x STM32f103C8T6 (Rev X) Blue pill boards (all modded with the 1K5 R10 resistor) . I flashed the bootloader on one of them in December 2017 and it works fine through USB. I tried to flash the current bootloader on the other 4 a couple of days ago and couldn't get any of them to be recognised on Windows 10. After messing around for a couple of days I dub out my backups and found the bootloader I installed back in 2017 and flashed that instead. With the old bootloader all four worked perfectly. 

I had a discussion with Ralph Bacon who has a YouTube channel (Ralph S Bacon) who had been having similar problems. I tracked down the older version of the Bootloader on this Github and sent him the link. It's the file dated 9 Aug 2017 - 
[https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/d19bcaf1ef044eb5e09b772f7a1d7c158c3c2976/binaries/generic_boot20_pc13.bin](url)

You can see the conversation in the comments of the following video -
[https://www.youtube.com/watch?v=vzDsSdtKQXA](url)

He installed this bootloader on his devices and reported that it also worked properly. He then tried the current bootloader on one of the new devices he had received directly from STM and reported that the current bootloader works fine on the newer chips.

I don't know if it helps but when I was trying to track down the cause of this problem I had my scope on the USB d+ line. When I looked at the signal on the first board with the old bootloader, I could clearly see and it was very obvious when the Bootloader pulled the D+ line low. On the other boards with the current bootloader I didn't see the D+ being pulled low at all (except for the low bits of the data of course)

bill-orange, 2019-12-11:
I appear to be having the same problem.  The link to the older 9 Aug 2017 firmware appears dead.  Can you post the firmware or a fresh link?

me21, 2019-12-28:
Well, if some older version of this bootloader works, and the new one doesn't, you can pinpoint the offending commit with git bisect.

rogerclarkmelbourne, 2019-12-28:
All binaries have been updated today, @dnfi should also just try today's update

bill-orange, 2019-12-28:
Where can I find the older versions?

me21, 2019-12-28:
This is what repository is for. Check out some earlier commit and build it.

rogerclarkmelbourne, 2019-12-29:
https://github.com/rogerclarkmelbourne/STM32duino-bootloader/commits/master

rogerclarkmelbourne, 2020-05-07:
There have been some bug fixes which may have resolved this problem

closed 2020-05-07 by rogerclarkmelbourne