rogerclarkmelbourne/STM32duino-bootloader #77: Bootloader for RobotDyn Blackpill STM32F303CCT6

pedrofurla, 2019-10-08:
Hi, I'd like to contribute a boot loader for this device, but I will need help figuring out how to even start. Any advices?

rogerclarkmelbourne, 2019-10-08:
The F303 is considerably different from the F103.

So a lot of changes would need to be made in relation to the interface to the USB subsystem in the IC.

There are other bootloaders e.g. the USB HID bootloader 

https://github.com/Serasidis/STM32_HID_Bootloader

Which has a F4 variant. Which may be easier to port to the 303 than this DFU bootloader

stevstrong, 2019-10-08:
The F303 USB part is very similar to F103 (and different from F4), so I think that porting the DFU bootloader from F103 would make a lot of sense.
I would recommend to read the USB spec of both devices and figure out the differences.

rogerclarkmelbourne, 2019-10-08:
@stevstrong 
Thanks for the correction.

stevstrong, 2019-10-08:
According to [this document](https://www.st.com/content/ccc/resource/technical/document/application_note/8e/c8/8d/e3/ee/ff/44/e6/DM00073522.pdf/files/DM00073522.pdf/jcr:content/translations/en.DM00073522.pdf) table 6 on page 12, the USB FS interface of STM32F303/302xB/C is fully compatible with STM32F1x.

The FLASH is however only partially compatible.

pedrofurla, 2019-10-08:
Thanks for the information, as it will apparently be a lot more work than I was expecting and the STM32F3 is on the boundary of my minimum requirements I decided to try the STM32F4DISCOVERY and see how it goes. 
I will keep this idea close by as I think it's nice hacking project though, and come back to it sometime.

Thanks for the very useful information!

stevstrong, 2019-10-09:
Roger, would it be possible to keep this open as I may have a look onto this in the near future.
I think the changes should only relate to flash handling, and that I assume to be not very complicated.

rogerclarkmelbourne, 2019-10-09:
Yep. No problem

rogerclarkmelbourne, 2020-05-07:
Closing because of no activity

stevstrong, 2020-05-08:
I just finished my CDC bootloader for F303CC which is based on the F1 version.
The changes are minimal, needs additional GPIO af setting beside setting mode, and change the flash size to 2kB.

rogerclarkmelbourne, 2020-05-08:
OK
Thanks

closed 2020-05-08 by rogerclarkmelbourne