rogerclarkmelbourne/STM32duino-bootloader #79: STM32F103RC "perpetual bootloader mode"

serkanc, 2019-12-06:
I have uploaded generic pc13 bootloader without any problem. Then I uploaded sketch with Arduino IDE, user code works fine, but after reset bootloader blinks led 6 times fast, then begins to blink slow, but don't jump to user code. I use my own hardware. What can be the problem. ?
Thanks .

mikegleasonjr, 2020-01-30:
I was able to upload sketches yesterday, but tonight the thing you describe happens. The sketch upload fine but the serial still shows the default bootloader message.

mikegleasonjr, 2020-02-04:
I was having the same problem, I was able to bring back the board to life by reflashing the bootloader, and pressing reset then immediately uploading a new sketch... sometimes it fails but pressing reset and then uploading a sketch it will at one point succeed.

Then I can proceed to upload sketched normally after. Weird.

rogerclarkmelbourne, 2020-05-07:
Closing because of no activity in the last 3 months

closed 2020-05-07 by rogerclarkmelbourne