rogerclarkmelbourne/STM32duino-bootloader #80: STM32FEBKx6 support?

arpruss, 2019-12-08:
Roger:

Do you have any idea why the bootloader doesn't work on the STM32FEBKx6 (found on some fake Blue Pill boards), which has 10K RAM and 32K flash? The USB hardware on these seems to be pretty much the same as on the STM32F103C8: I can use my USBComposite library with them with no changes.

rogerclarkmelbourne, 2019-12-08:
No.

Unfortunately, I don't think I have any of those devices.

I just opened a new BluePill which I received last week, and installed the bootloader and it seems to be working fine, so I guess they must be genuine.

Is there any way to tell which ones are fake just by at the markings.

If its only got 10k then the stack won't work, as AFIK the SP normally is initialised at the top of the available RAM

arpruss, 2019-12-08:
The markings look pretty genuine. But the Flash Demonstrator will say it's got 32K.

I just tried to change the _estack to 0x20002800 in the bootloader's ld file, and now the bootloader starts up and I can connect. But half way through an upload, it hangs, with the bootloader dead (presumably because the part of flash where the bootloader sits has been overwritten).

rogerclarkmelbourne, 2019-12-08:
The bootloader is at the bottom of flash, it should not be ovewrritten.

arpruss, 2019-12-08:
Does something maybe wrap around?

In any case, it all now works.  All I really had to do with the bootloader was lower estack from 0x20005000 to 0x20002800 in the ld scripts, and then correctly modify the ld scripts on the core side as well. Do you want to explore merging this in, or would the lower stack cause problems for some devices?

rogerclarkmelbourne, 2020-05-07:
The bootloader was updated a while ago to operate on devices with less RAM

closed 2020-05-07 by rogerclarkmelbourne