rogerclarkmelbourne/STM32duino-bootloader #83: Device is not Recognized on Windows 10 PC?

gunpal5, 2020-01-03:
Hello,

I am recently started working with STM32F103C8T6 blue pill board. I tried 3 different versions of usb bootloaders:
1) old version of generic_boot20_pc13.bin, 
2) latest generic_boot20_pc13.bin 
3) generic_boot20_pc13_fastboot.bin

I installed the drivers on my PC. but whenever I connect it to USB I get "Device not recognized". 

[image]

My STM32 board came with R10 = 10k. Is that causing this issue? or What else could be the reason for this? 

Any suggestion would help.

Regards
Gunpal Jain

hattesen, 2020-01-08:
1. You do realize, that this is NOT a support forum for general problems relating to STM32 development, right?

2. This issue tracker exists to report problem reports specific to the code contained in this repository, preferably with information of how to reproduce the problem.

3. Did you try to use Google, to see if anyone else has experienced the same error message on this type of hardware? If so, you would likely have run into support forum pages like this one: https://community.platformio.org/t/stm32duino-black-pill-usb-descriptor-failed/6040/5

gunpal5, 2020-01-08:
1) Yes, I know it is not a STM32 forum.
2) The problem I mentioned is related to this repository. I used the bootloader and drivers from this repository. I believe there's some issue in the windows driver on windows 10, but not exactly sure what. 
3) Yes, I tried to google this issue for several days. Thanks for the link, but I have made myself comfortable with the serial programmer and I gave up the hope of getting STM32 to work with USB.

Regards,
Gunpal Jain

rogerclarkmelbourne, 2020-05-07:
Try posting to www.stm32duino.com

closed 2020-05-07 by rogerclarkmelbourne