rogerclarkmelbourne/STM32duino-bootloader #85: BOOT1 high no longer forces bootloader mode in pre-built generic_boot20_pc13.bin

gabebuckmaster, 2020-01-22:
When I flash my board with the current version of generic_boot20_pc13.bin in bootloader_only_binaries, setting BOOT1 high while pressing reset does not cause the board to stay in bootloader mode.

Flashing my board with the version of generic_boot20_pc13.bin in bootloader_only_binaries from a16ff8e on Apr 20, 2018 causes it to work as expected.

gabebuckmaster, 2020-01-22:
I did some more testing, and this issue seems to have appeared in the most recent commit that affected the file (c235370). It works up through (and including) 125872b. I took a peek at the source but didn't see anything obvious that caused this.

rogerclarkmelbourne, 2020-01-23:
Looks like an error was introduced when I had to manually merge a change.

The line which checks where there application code to run, was inserted twice.

I think by removing the duplicate the bug has been fixed

gabebuckmaster, 2020-01-23:
I've confirmed that this fixes the bug. What service!

closed 2020-01-23 by gabebuckmaster

rogerclarkmelbourne, 2020-01-23:
No worries