rogerclarkmelbourne/STM32duino-bootloader #87: STM32F103C8 didn't change usb id after reset , stays on 0003 status .

SalimF, 2020-02-20:
Hi I'm not sure if this issue related to my OS `Linux mint 64bit` or to something I change in the board , 
the first board I l flashed using this bin file generic_boot20_pc13_fastboot.bin works well for two days bet from time time , 
I notice it's didn't showup on Arduino IDE under port menu , from my experience with the board its start when boot up at usb-id `1eaf:0003` and after few second it's changed to `1eaf:0003` which command `dmesg -T` show me Linux assess it to `/dev/ttyACM0` ,, but now the board only stuck on 0003 mode .
```
$ lsusb 
Bus 002 Device 082: ID 1eaf:0003
```
[image]

rogerclarkmelbourne, 2020-02-20:
The bootloaders VID:PID is 1eaf:0003

If the Arduino sketch has USB compiled in, it enumerates as 1eaf:0004

If you don't see  1eaf:0004 then either your sketch does not have USB in it (which will be compiled in with the "STM32duino bootloader" Upload option (and sever other Upload options)

Or your PC etc is not detecting the board has requested re-enumeration

This is not a bug in the bootloader, its either a hardware issue of a OS issue.

Note the Generic STM32F103 boards do not have the correct hardware to fore re-enumeration of the USB bus. 
We use a trick that seems to work on most machines but may not work with your PC.

Try posting to www.stm32duino.com or www.stm32world.com

closed 2020-02-20 by rogerclarkmelbourne

SalimF, 2020-02-21:
@rogerclarkmelbourne thank you very much for the valuable reply , this board (the blue pill) acts very weird with Arduino IDE , after I see your reply I start Arduino and hook-it up again and I jut hit upload button and it's work .

```
Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
```
here is again screenshot of my Arduino id menu setting it's may be the one that help finish uploading the sketch .
[image]

rogerclarkmelbourne, 2020-02-21:
Does the board have a jump link on Boot1 ?

Boot1 should be pulled low, as its used as a switch to lock the board into the bootloader, in case the sketch completely crashes and the USB stops working.

The IDE will try to upload even if it can't find the USB CDCACM

SalimF, 2020-02-21:
> Does the board have a jump link on Boot1 ?
yes i have and it's set to 0 same with Boot0
Anyway now everything work perfectly with same setting I post on the screenshot i manage to upload LED blink sketch on both builtin LED and PA0 and it's works . thanks again fro help and fast reply

taloot, 2020-05-16:
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

follow this