rogerclarkmelbourne/STM32duino-bootloader #93: STM32F1 not being recognized by windows with a 12MHz Crystal

ccccmagicboy, 2020-05-07:
I hope someone here might be able to help me out. My problem is when I connect up to the usb I get unrecognized device "Unknown USB Device (Device Descriptor Request Failed)". My board is a custom board with STM32F103RCT6 AND 12MHz crystal, and when burn the binaries from here, the board PC13 will blink a led, but without a successful usb 
[image]

rogerclarkmelbourne, 2020-05-07:
You will need to make a new build target as most other boards have a 8MHz crystal

See config.h section for TARGET_GD32F1_GENERIC_F103_PC13 which has a 12Mhz crystal

closed 2020-05-07 by rogerclarkmelbourne

ccccmagicboy, 2020-05-08:
Great! Thanks!!!

rogerclarkmelbourne, 2020-05-08:
No worries