rogerclarkmelbourne/STM32duino-bootloader #95: Compatibility with F303 Cpu

davidabevan, 2020-05-13:
Hi Roger, Thanks for your work.
I am a tinkerer not a programmer so hopefully posted in right place...
Question please, I have been home building custom pcb Bluepill clones from scratch and recycling my old F/C CPU,s. Have successfully built and programmed the f103 no probs. My second build uses the
stm32f303cct6, I cant at moment get the boot loader working, i use the generic PC13. I can upload using flash tool ok and verify also a blink sketch works ok so Am guessing either 1 most likely have some soldering or other issue with my board, currently investigated the obvious, or possibly the bootloader wont work on the stmf303 series, 
Your advice would be welcome.
Thanks 
Dave B..

rogerclarkmelbourne, 2020-05-13:
It only works on the F103 processor.

I think the USB HID bootloader (try googling for it) supports more processors, but I don't know if that includes the F3x series

closed 2020-05-13 by rogerclarkmelbourne

stevstrong, 2020-05-13:
The bootloader does not work on F3 blackpill (F303CCT6).
What is missing is:
- add AF mode setting to actual GPIO settings
- change FLASH page size to 2 kB.

Alternatively, you may consider the [CDC bootloader](https://github.com/stevstrong/STM32-CDC-Bootloader) which is adapted for F3 blackpill.

davidabevan, 2020-05-13:
Thanks stevstrong.. will give it a go

davidabevan, 2020-05-13:
Thanks this works So now at least i know my smd soldering working, thanks for the input..