rogerclarkmelbourne/STM32duino-bootloader #98: dfu-util doesn't see the device until I physically unplug it and plug again

bananu7, 2020-07-05:
Hi!

The board I have seems to be a variant of the "black pill", and it's labeled STM32F103C8T6.

Through trial and error I figured that it has LED on PB12, so I've been using the `generic_boot20_pb12.bin` file. I've installed the stm32duino drivers via the script, it does show in the device manager under "Maple DFU" with IDs `0x1EAF` and `0x0003`, however `dfu-util` only shows on the first bootloader start after power-up.

To program it, i was able to start the build in Arduino IDE, then plug the board once it started looking for devices; that went fine. If I start the board in perpetual bootloader, it also works fine and `dfu-util` can see it. However, once the board is soft-reset, it stops being visible. Nothing changes in the device manager, but `dfu-util` and thus my build cannot see it.

After uploading the first sketch, it shows up properly as `Maple Serial`, and after fixing the reset resistor (shipped with 10K) it's able to trigger the DTR soft reset, but that soft reset fails to put it in the DFU mode again.

I can confirm that the bootloader actually runs in all those cases because the diode blinks as expected; first quickly and then slowly for a while (or indefinitely in perpetual mode), but the device itself doesn't seem to respond properly.

I am not sure what's causing the problems, but since this is reproducible even without any code, just with the bootloader (it's enough to soft-reset the board to break it).

stevstrong, 2020-07-05:
Which OS and which core do you use?

bananu7, 2020-07-05:
Windows 10, the stm32duino. I've also tried from PlatformIO, but again, it fails (or rather requires unplug) on `dfu-util`; doesn't really matter what's on there, as it's reproducible with the board empty just with the bootloader in,

In fact, I've downloaded the sources and compiled my own binary based on PB12, but enabling the "maple reset" thing; I've no idea if that's what should be used, but it didn't help.

TBH I'm thinking that it might just be a fault of this particular black pill. It's _probably_ possible to make it soft-reset USB properly, but I have no idea how to even start figuring that bit out.

I think mine is exactly this one:
https://hobbycomponents.com/st-micro-electronics/960-stm32f103-black-pill-development-board

Got it from AliExpress, so can't be sure. Since then I've just moved to ST-link - so no bootloader. I can still try getting the bootloader to work, if you'd be interested in making this one work.

rogerclarkmelbourne, 2020-07-05:
I don't think this is an issue with the bootloader as it works fine with many different boards on Windows 7, 10 and Linux.

Fundamentally, most of these boards don't have a USB reset circuit as defined by the USB standard. The bootloader uses a hack/trick to reset the USB that works on most computers but not all.

I don't really think this is a place to discuss possible incompatibilities between your specific board and your PC.
I recommend you post this on www.stm32duino.com

bananu7, 2020-07-07:
Fair enough, although at least I was correct that it is, in fact, the bootloader failing to reset the USB on my board. As far as I'm concerned, I'm good; maybe I'll figure out how to reset the USB on this board but it's not a priority or a blocker for me right now.

closed 2020-07-10 by bananu7

piersh-aetheros, 2021-12-08:
I think I'm seeing the same issue. Directly after flashing with FTDI, I see the following in device manager:
[image]

After that, I can upload sketches built using PlatformIO using the following `platformio.ini`:

````
[env:bluepill_f103c8]
platform = ststm32
framework = arduino
board = bluepill_f103c8_128k
upload_protocol = dfu
board_build.core = maple
````

But at some point after uploading a sketch the Windows device manager shows this instead:
[image]

and I'm no longer able to upload sketches, even after power-cycling.

The only fix I know is to re-flash the bootloader.

Spongman, 2021-12-09:
I think this warning may be related:
```
Device USB\VID_1EAF&PID_0003\LLM_003 was not migrated due to partial or ambiguous match.

Last Device Instance Id: USB\Vid_80EE&Pid_CAFE\125D20140310
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Location Path: PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(9)#USB(1)
Migration Rank: 0xF000FFFFFFFF0023
Present: false
Status: 0xC0000719
```