rogerclarkmelbourne/STM32duino-bootloader #100: Can't put maple mini in perpetual bootloader mode

eagl1, 2020-08-12:
Hi,

Here's my full post in Arduino forum.

[https://forum.arduino.cc/index.php?topic=700036.msg4704325#msg4704325](url)

It was working ok on my PC and now I'm trying to upload a sketch on my laptop but it's not working.
1. Can't do the transfer to perpetual bootloader mode trick. The reset button and the "BUT" are doing the same 6 fast/slow blinks.

2. Can't upload any new sketch.

3. As in my Arduino forum post, hardly get it to be on port COM, after disconnect, it goes to Microchip tools category on windows device manager.

What is the problem ?

stevstrong, 2020-08-12:
Please use as reference the forum:
https://www.stm32duino.com

eagl1, 2020-08-12:
> Please use as reference the forum:
> https://www.stm32duino.com
but it's a read only forum, I can't post there !

stevstrong, 2020-08-13:
You need to register, as on each forum.
Here some readings: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki

closed 2021-11-09 by rogerclarkmelbourne