rogerclarkmelbourne/STM32duino-bootloader #102: PB2

tomgossland-cphatomics, 2020-11-25:
There's a new version of the bluepill that has its LED on PB2. I'm trying to make my way through the instructions to make a binary for this new board but I'm out of my element. Maybe someone who knows what they're doing would be able to do this faster? I'm not sure I've done it correctly and also struggling to run make on windows so I can't even test it.

tomgossland-cphatomics, 2020-11-25:
Nevermind!
https://github.com/WeActTC/BluePill-Plus/blob/master/SDK/STM32F103C8T6/Arduino/Bootloader/STM32duino-bootloader-PB2.bin

rohitsam, 2020-11-27:
I have made a wiki on how to compile this project  for your custom board [here](https://github.com/rohitsam/STM32duino-bootloader/wiki)

rogerclarkmelbourne, 2021-11-09:
OK

closed 2021-11-09 by rogerclarkmelbourne