rogerclarkmelbourne/STM32duino-bootloader #104: HID BOOTLOADER STM32F103R..

GitMauri, 2021-01-27:
Hi Thanks for your work and for your sharing.

I've used Serasidis Bootloader on Roger Clark core and it was working great on 103c8t6... now I've tried to use on STM32F103RCT6 and there are problems that I cannot fix...  (STM32 HID Bootloader 2.1.0)

I tried to add an edited hid_bootloader  file to ....\STM32F1\variants\generic_stm32f103r\ld      and I edited the boards.txt file for genericSTM32F103R  UPLOAD METHODS adding HID....

 I tried to recompile the generic-PD2.bin (make generic-pd2 PAGE_SIZE=2048) ... 
When I upload the sketch the IDE recognizes the board and it seems to upload correctly, but a simple blink doesn't work... (It works with Roger bootloader).
Maybe the HID bootloader is just not compatible with high density devices..? Can I use the board as a 128KB STM32F103RBT6? That should be a low density right? I tried but without luck...  I compiled on windows  using CYGWIN64 and the GNU Arm Embedded Toolchain: 10-2020-q4-major December 11, 2020.  It compiles without errors but  it gives some warning about deprecated things...

GitMauri, 2021-01-27:
Hi Roger

Thank you for your nice work during these years. It let me start playing with microcontroller for the first time...

I was thinking maybe other people didn't have this problem, and I've seen in your core the hid_bootloader.ld file is already inside the variant STM32F103C8T6, but not in other folders...

I'll try to write an ISSUE on Serasidis repo

Thanks again

Maurizio

rogerclarkmelbourne, 2021-01-27:
AFIK this repo does not have a file hid_bootloader.ld

GitMauri, 2021-01-27:
I just downloaded the ZIP file and it is inside...

Arduino_STM32/STM32F1/variants/generic_stm32f103c/ld/hid_bootloader.ld
Added support to HID Bootloader 2.0     3 years ago

GitMauri, 2021-01-27:
Ohh now I see, i did a mistake a put an issue on your bootloader and not on your core...  i did a mistake

rogerclarkmelbourne, 2021-01-27:
Ah. I was about to say, there are no Variants folder in this repo.

GitMauri, 2021-01-29:
Hi Guys

Maybe someone ended up here Googling, so it is useful to explain how I found a solution to my problems for the HID bootloader.
However I posted the issue here by mistake... I thought I was posting on the STM32 CORE repo.
Roger's bootloader works as good as a Swiss clock!

I used a mixed solution: I got the bootloader bin file from the new project of TheKikGen, but I used the Flasher (hidflash) of Serasidis, down I explain why.

KikGen also helped me to correct my wrong values in the ld file.

https://github.com/TheKikGen/stm32-tkg-hid-bootloader

I experienced some problems with the tkg-flash that KikGen has probably already fixed. Check out his new HID bootloader, very nice new features... he is tuning all up.

Here the chronology and details of my fix. 
 
https://github.com/Serasidis/STM32_HID_Bootloader/issues/44

Thanks to Roger, Serasidis, Arprus, Kikgen, Stevestrong, MrBurnette, FpiSTM and many others. who help us to have fun with our small projects without even checking one register... 

**Thanks to them we can focus on ideas and creativity forgetting the complexity...** 

Kudos, Sudos, Nudos, Pudos, Rados....  There are no words to express gratitude so I just forged some... :-)

closed 2021-01-29 by GitMauri