rogerclarkmelbourne/STM32duino-bootloader #108: Provide option to use HSI instead of the HSE clock

RoganDawes, 2021-05-25:
I have a commercial board (RF receiver) that I am trying to write a new/alternate firmware for. It would be nice to have a friendly bootloader, so that I don't have to use an external probe all the time. So far, however, I have been unable to get any of the example binaries to work.
The board uses an STM32F103RBT6, but it spins waiting for the HSE clock to start (as shown by various libopencm3 examples that use the HSE clock, unsuccessfully on this board).
It *is* capable of USB when using the HSI clock at 48MHz, however, even though I saw a comment comewhere that this is possibly "out of spec".

Would it be possible to provide an option to use only the HSI clock?

rogerclarkmelbourne, 2021-11-09:
This is not a bug.

I have no idea whether the STM32F103 supports USB using HSI. Probably not becuase HSI is not frequency stable.

closed 2021-11-09 by rogerclarkmelbourne

RoganDawes, 2021-11-10:
Hi Roger,

It may not be frequency stable, but it DOES work. STM32Duino USB-CDC is happy to run on this board, and I have not experienced any errors in the USB support yet. 

Unless somehow the Arduino core has a different way of starting the HSE clock that I am not aware of? As far as I am aware, the board I am using does not even HAVE an HSE clock. But STM32Duino "just works", with no tweaking required.