rogerclarkmelbourne/STM32duino-bootloader #109: Support for board "Olimexino STM32 (rev F)" ?

EvF, 2021-07-12:
Olimexino STM32 is a Maple-R5 clone but with extra connectivity on-board. Was working fine with old Maple IDE before I re-flashed it but does not work with the bootloader binaries/maple_rev5_boot20.bin

Programmed using a STM ST-Link V2 over SWD on Windows 10 x64 21H1
The board appears to start up OK but the USB COM device never appears.

I checked the board schematic and whilst it is slightly different (eg. DISC drives a PNP transistor rather than the 2x NPN) it all looks pretty much equivalent (LED1 is A5, BUT is C9, DISC is C12). MCU is ST32F103RBT6 same as the Maple board.

Schematic is here
https://www.olimex.com/Products/Duino/STM32/OLIMEXINO-STM32/resources/OLIMEXINO-STM32_Rev_F.pdf

Main board page is here
https://www.olimex.com/Products/Duino/STM32/OLIMEXINO-STM32

Any idea why this does not work?
TIA.

EvF, 2021-07-14:
It looks like the binary fails because it ends up in the middle of nowhere executing an illegal opcode.
I would like to debug this but when I try to build it under PlatformIO+vscode (which I figured was the way to go because there is a platformio.ini file in the repo) the build fails after several compiler warnings (including a warning of insufficient runtime stack space) and an unhelpful linker error1 (collect2)... it does not give any indication what symbols have failed to link.

Is there another build environment more likely to work?

rogerclarkmelbourne, 2021-11-09:
See you other issue

closed 2021-11-09 by rogerclarkmelbourne