rogerclarkmelbourne/STM32duino-bootloader #110: Not compiling in PlatformIO Core 5.1.1 Home 3.3.4

DrFrankReade, 2021-07-18:
Errors at compile time on every possible env. 

Probably very simple fix, I'm just a simpleton and it's beyond me.  See the compile logs below. 

Thank you!

```
Processing generic-pa1-button-pa8 (platform: ststm32; board: genericSTM32F103C8)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 (14.0.1) > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Warning! '-Wl,-T' option for specifying linker scripts is deprecated. Please use 'board_build.ldscript' option in your 'platformio.ini' file.
Compiling .pio\build\generic-pa1-button-pa8\src\dfu.o
Compiling .pio\build\generic-pa1-button-pa8\src\hardware.o
Compiling .pio\build\generic-pa1-button-pa8\src\main.o
Compiling .pio\build\generic-pa1-button-pa8\src\stm32_lib\c_only_startup.o
Compiling .pio\build\generic-pa1-button-pa8\src\stm32_lib\cortexm3_macro.o
Compiling .pio\build\generic-pa1-button-pa8\src\usb.o
Compiling .pio\build\generic-pa1-button-pa8\src\usb_callbacks.o
Compiling .pio\build\generic-pa1-button-pa8\src\usb_descriptor.o
dfu.c: In function 'dfuCopyState':
dfu.c:379:17: warning: return discards 'volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
         return (&(dfuAppStatus.bState));
                ~^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\generic-pa1-button-pa8\src\usb_lib\usb_core.o
In file included from usb_lib/usb_lib.h:21:0,
                 from usb.h:28,
                 from common.h:40,
                 from usb_descriptor.h:28,
                 from usb_descriptor.c:35:
usb_lib/usb_type.h:23:14: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
 #define NULL ((void *)0)
              ^
usb_descriptor.c:201:28: note: in expansion of macro 'NULL'
 u8 u8_usbStringInterface = NULL;
                            ^~~~
Compiling .pio\build\generic-pa1-button-pa8\src\usb_lib\usb_init.o
Compiling .pio\build\generic-pa1-button-pa8\src\usb_lib\usb_int.o
Compiling .pio\build\generic-pa1-button-pa8\src\usb_lib\usb_mem.o
Compiling .pio\build\generic-pa1-button-pa8\src\usb_lib\usb_regs.o
Linking .pio\build\generic-pa1-button-pa8\firmware.elf
c:/users/kontr/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\generic-pa1-button-pa8\firmware.elf section `._usrstack' will not fit in region `RAM'
c:/users/kontr/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: section .init LMA [08001910,08001913] overlaps section .data LMA [08001910,0800200b]
c:/users/kontr/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 1097 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\generic-pa1-button-pa8\firmware.elf] Error 1
================================================================================================================================ [FAILED] Took 2.16 seconds ================================================================================================================================
```

juergs, 2021-07-18:
Not really a bootloader-problem.
Take a look there https://github.com/platformio/platform-ststm32/issues.

EvF, 2021-07-21:
Same problem here (I mentioned over in issue #109).
Trying to build any bootloader targets fails with same linker errors.
Are you saying it's a platformIO issue? I didn't see it over in the platform-ststm32 issues list.
Is there a workaround?

Linking .pio\build\generic-pa1-button-pa8\firmware.elf
c:/users/andy/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none eabi/bin/ld.exe: .pio\build\generic-pa1-button-pa8\firmware.elf 
section `._usrstack' will not fit in region `RAM'
c:/users/andy/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: section .init LMA [08001910,08001913] overlaps 
section .data LMA [08001910,0800200b]
c:/users/andy/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 1097 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\generic-pa1-button-pa8\firmware.elf] Error 1

rogerclarkmelbourne, 2021-11-09:
This bootloader build from bare metal gcc

If you want to build using third party tools like Platformuino, you would need to modify the code yourself so that it built under that envirmonment

closed 2021-11-09 by rogerclarkmelbourne