rogerclarkmelbourne/STM32duino-bootloader #116: Which one of the binaries should I use for the mini stm32F103ZET6 chip with 144 pin package ?

eagl1, 2022-01-19:
Hi,

I have this board:

[image]

I found this video, and thought maybe this method would work to flash the board.

I downloaded these software:

1. STM32 Flash Loader
2. STM32 USB Mini Driver
3. STM32duino Bootloader

So the link of point 3 directed me to this page and found many bin files, which one should I download for the chip in this board ?

Also, if there are other steps regarding programming this board, please tell me what to do exactly ?

203Null, 2022-01-19:
You can use the PE2 one which will light up the user2 led

eagl1, 2022-01-19:
I couldn't connect it on Arduino IDE

I installed ST32CubeProgrammer with UART method but the same problem, I get this result:

[image]

eagl1, 2022-01-19:
Setting boot1 to GND I got this:

[image]

rogerclarkmelbourne, 2022-10-29:
Marked as solved by OP

closed 2022-10-29 by rogerclarkmelbourne

eagl1, 2022-10-29:
Thanks Roger, it's working very well with the ST LINK V2.