rogerclarkmelbourne/STM32duino-bootloader #117: dfu-util can't see DFU-bootloader at STM32F103C8T6

xintrea, 2022-04-09:
Hello!

I am trying write **generic_boot20_pc13.bin** to my STM32F103C8T6 board by ST-LinkV2:
```
st-flash --reset write generic_boot20_pc13.bin 0x08000000
```
After push Reset, the board showing in **dmesg**:
```
[25760.232130] usb 2-2: new full-speed USB device number 7 using xhci_hcd
[25760.385437] usb 2-2: New USB device found, idVendor=1eaf, idProduct=0004, bcdDevice= 2.00
[25760.385444] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[25760.385447] usb 2-2: Product: Maple
[25760.385449] usb 2-2: Manufacturer: LeafLabs
[25760.424009] cdc_acm 2-2:1.0: ttyACM0: USB ACM device
[25760.424307] usbcore: registered new interface driver cdc_acm
[25760.424310] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```
But dfu-util can't see any devices:
```
dfu-util --list
dfu-util 0.9
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
```
I am truing run dfu-util with device id:
```
dfu-util -d 1eaf:0004 --list
```
But already nothing.

Why this DFU-bootload can not see at dfu-util?

xintrea, 2022-04-09:
Aha, author can't update doc/readme from user issue:

https://github.com/rogerclarkmelbourne/STM32duino-bootloader/issues/90

In this issue says:

> If a button pin is high (e.g. button pressed) at startup, the bootloader also waits for an upload indefinitely (even for fastboot). This can be used to force bootloader mode, even when the main application is broken and fastboot is enabled. The «button» pin varies per config (see config.h), but the led-on-PC13 variant that is typically used for the Blue Pill boards ahs the «button» pin configured to PB2, aka BOOT1, which is available on a pin header. The BOOT1 pin is normally used by the hardware to decide between system flash and RAM, but only when BOOT0 is 1. When BOOT0 is 0, it always boots from main flash, ignoring the value of BOOT1, so it can be used by the bootloader.

Summary: for set DFU-bootloader to DFU-ready mode, set boot jumpers to official undocumentet position boot0=0, boot=1.

After this setting, board shows as:
```
[ 4304.984940] usb 2-2: new full-speed USB device number 35 using xhci_hcd
[ 4305.133726] usb 2-2: New USB device found, idVendor=1eaf, idProduct=0003, bcdDevice= 2.01
[ 4305.133733] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4305.133735] usb 2-2: Product: Maple 003
[ 4305.133738] usb 2-2: Manufacturer: LeafLabs
[ 4305.133739] usb 2-2: SerialNumber: LLM 003
```
And DFU device list getting is work:
```
> dfu-util --list
Found DFU: [1eaf:0003] ver=0201, devnum=35, cfg=1, intf=0, path="2-2", alt=2, 
           name="STM32duino bootloader v1.0  Upload to Flash 0x8002000",
           serial="LLM 003"

Found DFU: [1eaf:0003] ver=0201, devnum=35, cfg=1, intf=0, path="2-2", alt=1,
           name="STM32duino bootloader v1.0  Upload to Flash 0x8005000",
           serial="LLM 003"

Found DFU: [1eaf:0003] ver=0201, devnum=35, cfg=1, intf=0, path="2-2", alt=0,
           name="STM32duino bootloader v1.0  ERROR. Upload to RAM not supported.",
           serial="LLM 003"
```
Details in the article: 

https://webhamster.ru/mytetrashare/index/mtb0/1645994134zdp61ftj33 (RU)

rogerclarkmelbourne, 2022-10-29:
This is not a issue with the code.

Try www.stm32duino.com

closed 2022-10-29 by rogerclarkmelbourne