rogerclarkmelbourne/STM32duino-bootloader #118: I have a bin file, what is the proper procedure to flash my generic stm32f103c chip?

dleung18, 2022-06-07:
I have already flashed the bootloader in the chip and it is seen by my computer as maple mini comport. It works fine in Arduino IDE.
Should I flash the bin file beginning from the flash memory location x800200 , just after the boundary of the 8k bootloader?

dleung18, 2022-06-09:
Here is the console command, I ran the dfu-utils.
D:\Arduino\hardware\Arduino_STM32\tools\win\dfu-util-0.9-win64>dfu-util.exe --list
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [1eaf:0003] ver=0201, devnum=13, cfg=1, intf=0, path="2-1", alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000", serial="LLM 003"
Found DFU: [1eaf:0003] ver=0201, devnum=13, cfg=1, intf=0, path="2-1", alt=1, name="STM32duino bootloader v1.0  Upload to Flash 0x8005000", serial="LLM 003"
Found DFU: [1eaf:0003] ver=0201, devnum=13, cfg=1, intf=0, path="2-1", alt=0, name="STM32duino bootloader v1.0  ERROR. Upload to RAM not supported.", serial="LLM 003"

D:\Arduino\hardware\Arduino_STM32\tools\win\dfu-util-0.9-win64>dfu-util.exe -a 1 -s 0x08005000:leave -D EMBO_F103C8.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #1 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Could not read name, sscanf returned 0
Failed to parse memory layout

D:\Arduino\hardware\Arduino_STM32\tools\win\dfu-util-0.9-win64>

Communication timeout error was encountered when the pc application connecting the comport.

Any ideas?

dleung18, 2022-06-10:
Remove the switch -s , file is being downloaded, but got win error, USB was not reset.
D:\Arduino\hardware\Arduino_STM32\tools\win>dfu-util -a 2 -D EMBO_F103C8.bin
dfu-util - (C) 2007-2008 by OpenMoko Inc.
This program is Free Software and has ABSOLUTELY NO WARRANTY

Opening USB Device 0x0000:0x0000...
Found Runtime: [0x1eaf:0x0003] devnum=1, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Setting Configuration 1...
Claiming USB DFU Interface...
Setting Alternate Setting ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
Transfer Size = 0x0400
bytes_per_hash=1043
Starting download: [##################################################] finished!
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!

Then, windows always see it as a Maple UDF, no more comport in device manager.
I think the USB descriptor code 0xleaf:0x0003 would not fall back to 0xleaf:0x0004.

Any suggestions to get it fixed? re-compile the whole codes?

dleung18, 2022-06-11:
I also use the maple_upload.bat and got the same thing. No comport showed up even USB device not recognized in windows device manager.  Here is the console capture. D:\Arduino\hardware\Arduino_STM32\tools\win>maple_upload.bat COM7 2 1EAF:0003 EMBO_F103C8.bin
maple_loader v0.1
Resetting to bootloader via DTR pulse
Reset via USB Serial Failed! Did you select the right serial port?
Assuming the board is in perpetual bootloader mode and continuing to attempt dfu programming...

Searching for DFU device [1EAF:0003]...
Found it!

Opening USB Device 0x1eaf:0x0003...
Found Runtime: [0x1eaf:0x0003] devnum=1, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Setting Configuration 1...
Claiming USB DFU Interface...
Setting Alternate Setting ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
Transfer Size = 0x0400
bytes_per_hash=1043
Starting download: [##################################################] finished!
error resetting after download: usb_reset: could not reset device, win error: A device which does not exist was specified.

state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode

timeout waiting for COM7 serial
D:\Arduino\hardware\Arduino_STM32\tools\win>maple_upload.bat COM7 2 1EAF:0003 EMBO_F103C8.bin
maple_loader v0.1
Resetting to bootloader via DTR pulse
Reset via USB Serial Failed! Did you select the right serial port?
Assuming the board is in perpetual bootloader mode and continuing to attempt dfu programming...

Searching for DFU device [1EAF:0003]...
dfu-util - (C) 2007-2008 by OpenMoko Inc.
This program is Free Software and has ABSOLUTELY NO WARRANTY

Couldn't find the DFU device: [1EAF:0003]
^CTerminate batch job (Y/N)? y

D:\Arduino\hardware\Arduino_STM32\tools\win>

rogerclarkmelbourne, 2022-10-29:
This is not an issue with the code.  Try www.stm32duino.com

closed 2022-10-29 by rogerclarkmelbourne