rogerclarkmelbourne/STM32duino-bootloader #120: can this bootloader use in STM32F103RCT6?

mcykin, 2022-11-26:
i have a develope board which use STM32F103RCT6,can i download this bootloader?

rogerclarkmelbourne, 2022-11-26:
Bootloader will work with the F103RCT but every board uses different GPIO pins for the button used to enter the bootloader.

Look in https://github.com/rogerclarkmelbourne/STM32duino-bootloader/tree/master/bootloader_only_binaries  for a binary which has the same pin as your board uses for its bootloader button.

If you don't find a precompiled binary that matches your board you'll need to modify and compile the code to suit your board

closed 2022-11-26 by rogerclarkmelbourne

mcykin, 2022-11-27:
hi,i've check my developed board,but i find out it didn't have any button except NRST pin.

[image]

as you can see,this board only leave a RST switch and BOOT0/1 select pin.

so,which version of boodloader should i use? can i still use binaries/generic_boot20_pc13.bin to download?

rogerclarkmelbourne, 2022-11-27:
Try asking on https://stm32duino.com

Fusseldieb, 2023-08-21:
> hi,i've check my developed board,but i find out it didn't have any button except NRST pin.
> 
> [image]
> 
> as you can see,this board only leave a RST switch and BOOT0/1 select pin.
> 
> so,which version of boodloader should i use? can i still use binaries/generic_boot20_pc13.bin to download?

I have the same board. Did you manage to get it working? What did it need?

DFWJunior, 2023-08-26:
> > so,which version of boodloader should i use? can i still use binaries/generic_boot20_pc13.bin to download?
> 
> I have the same board. Did you manage to get it working? What did it need?

i think i find the way to use it.
so you need a usb-ttl device,it can find in many ways.

first,put BOOT0 switch to 1(3.3v), BOOT1 should be fine in default.
and you need to connect your TTL downloader to RCT6 int these ways:

3.3V -- 3.3V(Careful your TTL Voltage!)
TX -- A10
RX -- A9
GND -- GND

then it should be fine to download it,like,i'm using STM32's Demonstration GUI to downloade stm32duino bootloader to use QMK function.