rogerclarkmelbourne/STM32duino-bootloader #122: upload via platformio

tcurdt, 2023-02-25:
This is what I tried:

```
$ st-info --probe
Found 1 stlink programmers
  version:    V2J17S4
  serial:     54FF67064882565217411887
  flash:      65536 (pagesize: 1024)
  sram:       20480
  chipid:     0x0410
  descr:      F1xx Medium-density
$ st-flash write generic_boot20_pc13.bin 0x8000000
...
2023-02-25T17:29:37 INFO common.c: Flash written and verified! jolly good!
```

Now when I connect to the usb serial I the following message in a loop:

```
Congratulations, you have installed the STM32duino bootloader
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

For more information about Arduino on STM32
See https://www.stm32duino.com
```

That looks like the bootloader and the welcome sketch works correctly.

I now change the `platformio.ini` to

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = dfu
```

but when running `pio run -t upload` I am getting an error (which seems to be related to an outdated dfu-util)

```
dyld[54773]: Library not loaded: /opt/local/lib/libusb-1.0.0.dylib
  Referenced from: /Users/***/.platformio/packages/tool-stm32duino/dfu-util/dfu-util
  Reason: tried: '/opt/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file)
/Users/***/.platformio/packages/tool-stm32duino/maple_upload: line 53: 54773 Abort trap: 6           ${DFU_UTIL} -d ${usbID} -a ${altID} -D ${binfile} -R ${dfuse_addr} -R
```

Updating it `brew install dfu-util` and changing the reference in `~/.platformio/packages/tool-stm32duino/maple_upload`, the upload succeeds:

```
...
Download done.
DFU state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Resetting USB to switch back to runtime mode
```

but after that upload the bootloader seems to be gone again. No usb serial shows up anymore.

I then also tried

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = dfu
board_upload.offset_address = 0x2000
```

Since the bootloader was working, this is rather a platformio issue, right?

But is `dfu` the right upload method to use with the bootloader?

It just has to be written at the correct address? Which is 0x8002000?

While digging through some github issues I also found these settings:

```
board_build.core = maple
build_flags = -DSERIAL_USB -DGENERIC_BOOTLOADER
```

Just not sure whether they make sense or not.

I would be very thankful for some input.

rogerclarkmelbourne, 2023-02-25:
This bootloader was written by LeafLabs to work with the Arduino IDE.

Try posting to www.stm32duino.com to see if someone can help you make it work with another IDE

closed 2023-02-25 by rogerclarkmelbourne

tcurdt, 2023-02-25:
I will - but could you just confirm how it should work in general?

What's the mechanism to upload and the address is for the sketch is 0x8002000?

kr4fty, 2024-07-21:
> I now change the `platformio.ini` to
>
> ```
> [env:bluepill_f103c8]
> platform = ststm32
> board = bluepill_f103c8
> framework = arduino
> upload_protocol = dfu
> ```
>
> but when running `pio run -t upload` I am [...]

OS: Linux Manjaro
IDE: VSCode (VSCodium)
PlatformIO: Core v6.1.15  Home v3.4.4

I have it working and this is my platformio.ini configuration file

```
[env:genericSTM32F103C8] ; Usando bootloader generic_boot20_pc13.bin
platform = ststm32
board = bluepill_f103c8
board_build.mcu = stm32f103c8t6
framework = arduino
upload_protocol = dfu
upload_command = dfu-util -a 2 -D $SOURCE
```
Then when i do an Upload the firmware:

```
Checking size .pio/build/genericSTM32F103C8/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.5% (used 1132 bytes from 20480 bytes)
Flash: [==        ]  16.8% (used 10992 bytes from 65536 bytes)
Building .pio/build/genericSTM32F103C8/firmware.bin
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: /dev/ttyS31
Uploading .pio/build/genericSTM32F103C8/firmware.bin
dfu-util 0.11

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 1eaf:0003
Device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Interface #2 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device

Download        [                         ]   0%            0 bytes
Download        [==                       ]   8%         1024 bytes
Download        [====                     ]  16%         2048 bytes
Download        [======                   ]  24%         3072 bytes
Download        [========                 ]  33%         4096 bytes
Download        [==========               ]  41%         5120 bytes
Download        [============             ]  49%         6144 bytes
Download        [==============           ]  58%         7168 bytes
Download        [================         ]  66%         8192 bytes
Download        [==================       ]  74%         9216 bytes
Download        [====================     ]  83%        10240 bytes
Download        [======================   ]  91%        11264 bytes
Download        [======================== ]  99%        11284 bytes
Download        [=========================] 100%        11284 bytes
Download done.
```