rogerclarkmelbourne/STM32duino-bootloader #131: How to flash a custom STM32F103C8 board?

etale-cohomology, 2024-08-15:
I have a custom STM32F103C8 board that has BOOT0 connected to ground (no resistor), NRST connected to ground (via a 100nF capacitor) and also to a pad, and PB2 floating.
PA12 (USB D+) is pulled high via a 1.5K Ohm resistor.
I'm using a 16 MHz crystal.
There are no LEDs connected to any GPIO pins.
I've added `udev` permissions for my ST-LINK programmer and for `1eaf:0003`.

I can flash `.bin` programs into the STM32 and they work.
I can also flash this bootloader into it.

I've tried (recompiling and flashing) `generic-none` and `generic-pc13`, among others.
I added `XTAL16M` as a `#define` before recompiling.

For `generic-none`, `lsusb` shows nothing (device doesn't seem to be detected at all) and can't DFU-flash anything.

For `generic-pc13`, `lsusb` does detect the device, showing something like: `Bus 003 Device 010: ID 1eaf:0003 Leaflabs Maple DFU interface`.
Moreover, I can use this bootloader to copy a program into the STM32 using `dfu-util`, eg.:

```sudo dfu-util -a 2 -d 1EAF:0003 -D myfirmware.bin```

**However**, at this point things stop working. Even after reconnecting the STM32 board, the entry for `Bus 003 Device 010: ID 1eaf:0003 Leaflabs Maple DFU interface` appears but never disappears, and my firmware never seems to run.

I have another custom STM32F103 board that I flashed/DFU'd with this bootloader successfully, and I know that first you get a USB entry for `1eaf:0003 Leaflabs Maple DFU interface` and afterwards you get a USB entry for your own firmware, but that's not happening in this case. It's as if the device is never leaving DFU mode and my firmware doesn't get to run.

etale-cohomology, 2024-08-15:
I can flash any of the STM32duino-bootloader `.bin` files into the STM32, but some are not detected in `lsusb`, some are detected but I can't DFU-flash anything, and for the rest I can DFU-flash my firmware but it doesn't get to run.
Here's a summary of the versions that work until the DFU-flash stage.

Works:

```
generic_boot20_pc13.bin
generic_boot20_pa9.bin
generic_boot20_pb0.bin
generic_boot20_pb7.bin
generic_boot20_pb9.bin
generic_boot20_pc13.bin
generic_boot20_pd1.bin
generic_boot20_pd2.bin
generic_boot20_pe2.bin
generic_boot20_pe5.bin
generic_boot20_pe5_button_pa0.bin
generic_boot20_pg15.bin
generic_boot20_pa1.bin
generic_boot20_pa1_button_pa8.bin
```

Works not:

```
generic-none_bootloader.bin
generic_boot20_pc13_fastboot.bin
```

etale-cohomology, 2024-08-19:
I was given the solution.
Praise the Lord Jesus.
I'm closing this issue.

closed 2024-08-19 by etale-cohomology

hdwong, 2025-03-07:
I want to know how it was resolved, as I am currently facing a similar situation. thanks